gnea/grbl #181: Excessive line gcode torture test

nathanielstenzel, 2017-04-28:
I have had problems lately with my grbl + arduino uno rev 3 combination lately. I had updated the grbl firmware to 1.1f and the usb interface to the latest version. I am still having trouble though. By any chance could someone test https://github.com/nathanielstenzel/bucket_mill/blob/960e67b3c1161c8d99741c011e9ab1027d35c0be/example_output/owl-trace-A3mm.gcode and see if it works for them? It has consecutive lines where the only movement was in the same direction as the last movement and it only moved 1mm.  Yes, that is far from efficient and I have that fixed in the program that I was working on that made the gcode. I wanted to bring this up though in case there is an actual problem with the position tracking in grbl. You see, while cutting away, it suddenly ends up at a deeper position than the gcode instructs it to go to. It could be that it can complete the movement on the gcode line faster than it can keep track of the position, but that is only a theory.

X3msnake, 2017-04-28:
> 960e67b3c1161c8d99741c011e9ab1027d35c0be/example_output/owl-
> trace-A3mm.gcode and see if it works for them? It has consecutive lines

Very Small and fast movements are a recipe to motor slipping and skipping
steps.

What motors and drivers are you using and what microstepping configuration
os set?

nathanielstenzel, 2017-04-28:
The reported x,y,z position ends up suddenly wrong which is why I suspect that it could be grbl that has the issue. 
gshield v5b
Arduino Uno R3
nema 17 motors
Since the position was not reported as being where it was supposed to be, I do not think there is a problem with skipped steps unless the grblshield can detect those, which I doubt. If it could, it would surely just correct for the skipped steps.

nathanielstenzel, 2017-04-28:
I am also careful with my acceleration settings.

nathanielstenzel, 2017-04-28:
Well, I did finally have some skipped steps...probably due to resistance, but that is still a separate issue.

chamnit, 2017-04-28:
@nathanielstenzel : If Grbl is reporting the correct position, then its skipped steps. Grbl tracks position by counting the step pulses while it executes them. You can even check this with a pulse counter sketch with a separate Arduino if you want. 

It would also help to have some information about your setup and machine. Provide your $$ Grbl settings, $I build info strings, and some details about your machine. 

Again, there are tens of thousands of Grbl machines deployed by several companies. None of them report issues with lost steps, except when the Grbl settings are configured as too aggressive. If you back off the acceleration and max rate and decrease your junction deviation (0.01mm or less), you should see skipped steps go away.

nathanielstenzel, 2017-04-28:
@chamnit There were times when there were no skipped steps, it reported Z-23 and the depth cut was supposed to be Z-10. Nowhere in the gcode says Z-23.
)��
Grbl 1.1f ['$' for help]
$$
$I
$0=10
$1=25
$2=100
$3=0
$4=0
$5=0
$6=0
$10=1
$11=0.010
$12=0.002
$13=0
$20=0
$21=0
$22=0
$23=0
$24=25.000
$25=500.000
$26=250
$27=1.000
$30=1000
$31=0
$32=0
$100=40.000
$101=40.000
$102=320.000
$110=2000.000
$111=2000.000
$112=2000.000
$120=50.000
$121=50.000
$122=50.000
$130=200.000
$131=200.000
$132=200.000
$
 $$
$0=10
$1=25
$2=100
$3=0
$4=0
$5=0
$6=0
$10=1
$11=0.010
$12=0.002
$13=0
$20=0
$21=0
$22=0
$23=0
$24=25.000
$25=500.000
$26=250
$27=1.000
$30=1000
$31=0
$32=0
$100=40.000
$101=40.000
$102=320.000
$110=2000.000
$111=2000.000
$112=2000.000
$120=50.000
$121=50.000
$122=50.000
$130=200.000
$131=200.000
$132=200.000

nathanielstenzel, 2017-04-28:
I changed the Z settings as follows just now for the next test. I also put a little bit of lubricant ont he Z bearing for my Shapeoko2.
$112=1500.000
$122=25.000

nathanielstenzel, 2017-04-28:
Could counting the X,Y movement cause it to get behind on its Z movement and then catch up on the Z movement and find itself at an odd position?

electrokean, 2017-04-28:
The only way I'd expect this issue is if you were mixing up G90/G91 modes, but it doesn't look like it.
And I note that your cut goes down to Z-14, not Z-10 - quite a way off Z-23, but still...
I've just moved office so I don't have any gear set up to actually try this, but the g-code seems to view in NCPlot fine.

chamnit, 2017-04-28:
@nathanielstenzel : Your settings don't look unusual, except your Z-axis steps/mm are a little high. If you are getting a Z-axis loss, then it's likely you are losing steps during a plunge motion. I've seen problems with lost steps when a CAM tool path will do a helical plunge or a ramp into material. This type of motion requires a lot of torque from the z-axis. The solution has been to lower your z-axis acceleration and max-rate and increase your micro stepping to half or full steps. The latter will improve the holding torque and should help reduce lost steps.

Grbl does not lose sync of motions, otherwise it wouldn't be CNC. Grbl is specifically written to never have this problem.

chamnit, 2017-04-28:
@nathanielstenzel : Sorry, misread your response. If Grbl is showing a value that is not in the program, then it's likely something is getting lost in the transmission between Grbl and the host UI. I don't see anything out of the ordinary with your g-code program. Laser raster engraving jobs do similar type of motions, but at 0.2-0.4mm increments and much faster block transmission rates. 

You may be getting fairly close to the max performance of Grbl in terms of g-code blocks/sec. I've tested the Uno does about 250 blocks/sec reliably. Certain configurations and jobs have been reported to exceed 1000 blocks/sec (laser rasters).  If Grbl runs ok if you lower the feed rate of your job by half, that would provide a clue where the problem is. Either Grbl hits its performance limits or comm is too fast for your UI.

chamnit, 2017-04-28:
Running the program now based on your $$ settings you posted. You are no where near the execution limits of Grbl. The program is relatively slow. 

I would try using Grbl's stream.py script or UGS/bCNC/Grbl Panel. If still see the error, then this would rule out your UI and is likely a setting or binding/friction.

nathanielstenzel, 2017-04-28:
@electrokean when it went to Z-23 or whatever, it was supposed to be at Z-10. It did not make it as far as the Z-14 section of the gcode.
@chamnit thanks, I will look up that script. I need reliable alternatives to my usual CNC interfacing software. Z is on a threaded rod while X and Y are on a belt. What "blocks" do you mean per second? Could high steps/mm plus fast Z movement cause issues?

Thanks all for your assistance with this so far.

nathanielstenzel, 2017-04-28:
Oh yeah. Please note that the link that I provided above should be for a specific version of that file. The torture test aspect comes in line sequences like X23, X24, X25, X26, Y43, X27, X28, etc. It is something that I should not have allowed to happen in the gcode and is now fixed. That inefficiency, as I mentioned, I figure might be useful for testing. Come to think of it, maybe that would be called extra "blocks" like @chamnit was talking about?

chamnit, 2017-04-28:
@nathanielstenzel : A line of g-code is called a block.

I've been running the test program. Nothing wrong so far after a few runs.

nathanielstenzel, 2017-04-28:
@chamnit do you have more or less steps/mm and mm/second than me with your settings?

X3msnake, 2017-04-28:
what UI sender are you using?

chamnit, 2017-04-28:
@nathanielstenzel : I'm running with your $$ settings, like I said before.

nathanielstenzel, 2017-04-28:
@X3msnake chilipeppr.com/jpadie is what I was using.
@chamnit  Oh. Okay. Thanks.

X3msnake, 2017-04-28:
What is the Machine architecture that is sending the code

PC Raspi, What browser are you using?

Do you have a way to try sending the code with another sender like UGS  -
https://winder.github.io/ugs_website/

electrokean, 2017-04-28:
@nathanielstenzel you mentioned running grbl on an "Arduino Uno R3"
Have you read the known issues section regarding the USB to serial converters, i.e. the 16U2 chip on the R3, or the CH340 on some clones
See https://github.com/gnea/grbl/wiki/Known-Issues#usb-to-serial-transmission-errors
Maybe you've managed to trigger this.  As @chamnit indicates it seems like you may be having communications errors causing some lost characters or similar - either that or a problem with your GUI streaming.
Long background discussion at https://github.com/grbl/grbl/issues/845

chamnit, 2017-04-28:
@nathanielstenzel : Chilipeppr has known performance issues. It struggles with high block rates. It's completely possible that the data from the web interface to the web server and onto Grbl is getting truncated or corrupted. Always try to use another GUI to determine if the problem is GUI or Grbl related.

closed 2017-04-28 by chamnit

nathanielstenzel, 2017-04-28:
@X3msnake I am using Chilipeppr/jpadie from a Chrome browser on a laptop running Fedora.
@electrokean I updated the USB firmware of the arduino. That is where you were going with that, right? 
@chamnit yes, I will try one of he other programs you guys mentioned.