gnea/grbl #182: Resetting Mpos coordinates

ghartc, 2017-04-28:
Hi - I am looking for a command/method to change MPosition coordinates - e.g. M position has some non-zero values, and I would like to zero them - x0y0z0 w/o moving the machine. Thanks.

chamnit, 2017-04-28:
@ghartc : Grbl intentionally does not allow this due the high likelihood of misuse and would make homing useless. You will want to learn how to use work coordinate systems to set multiple origins in your machine.

closed 2017-04-28 by chamnit

Harvie, 2018-11-21:
I am not sure if that is reliable approach, but when i do hard reset, it also resets my MPos...

Harvie, 2018-11-21:
It is important for me, because i have machine without limit switches, but i still occasionaly need to debug g-code using G53...

chamnit, 2018-11-21:
@Harvie : This is a very old thread you've resurrected. The answer is the same.

Harvie, 2018-11-21:
I know. I fully agree with fact that people should not be allowed to reset MPos.
But just for the record i've added that there is legitimate reason to do so and that it can be done by hard reseting if someone ever needs it.

Will hard reset always reset MPos? Or there are some cases in which it will fail?

chamnit, 2018-11-21:
A hard reset will always reset MPos.