gnea/grbl #183: is there an easy way to support non-standard geometry machines?

davidelang, 2017-04-29:
I'm looking at the maslow cnc machine ( https://www.kickstarter.com/projects/1830738289/maslow-cnc-a-500-open-source-4-by-8-foot-cnc-machine ) which has a very non-standard geometry (vertical machine, motors/encoders on the top corners with the router suspended from the two motors).

The project has a very primitive g-code interpreter that it's written. I was wondering if there is an easy way to marry the kinematics code that moves the hardware that they have written with the much more mature and robust g-code handling, interpretation, and movement planning that grbl has.

due to the mechanics of the maslow, the actual motion handling is very different, so I'm hoping that there are just a few functions that would have to be overridden. the maslow is similar to a delta printer, but works by adjusting the length of the links rather than the vertical location of the ends of the links. It also uses a closed loop system instead of steppers (h-bridge to control the speed/direction of the motors and then an encoder to detect the position of the shaft)

I've done a little digging in the code and traced the command execution down through the planner, but I'm not sure what processes the results of the planning to turn it into actual machine movement. It looks like it may be in stepper.c but I'm not sure.

winder, 2017-04-29:
Check out the COREXY flags for how different kinematics can be implemented.

davidelang, 2017-04-29:
Ok, I think I see how the conversion would work. One headache is that it is not a linear conversion. adjusting one motor doesn't move the tool in a straight line, even setting both motors moving at a given rate isn't going to move it in a straight line (except for a few special cases). In fact, moving 1mm in a given direction will require different movement depending on where you are in the workspace at the start of the movement (see http://2e5.com/plotter/V/design/ for a simplified version)

so you can't plan out large moves as 'move at a rate of X steps/sec on motor 1 and Y steps/sec on motor 2', you have to iterate through 'I have X mm to move, I'll do it in Y steps, and each step requires a calculation before knowing what to do for the next step.

Then combine this with closed-loop control (apply 10% power to motor 1 and 15% to motor 2 and then if motor 1 is moving too fast, apply less power to it), things get really "interesting" :-)

Doing a bit more digging, I'm thinking that the right answer is to leave the step calculations as they are and then replace the motor control (that it looks like is done in stepper.h/c) with completely different code that treats the desired step rates as a series of calculated positions and 'chases' them to minimize the error between the actual position and the calculated position.

does this sound reasonable?

chamnit, 2017-05-30:
Grbl assumes axes that are not dynamically coupled (i.e. cartesian). The Maslow is not. You can get it to work with Grbl's motion planner, but it won't be optimal. That said, the Maslow is not very efficient at cutting aggressively. It has to be weighed down by two large bricks (literally) to keep it down. So you'll never get to the upper end of performance that requires an optimal planner.

closed 2017-05-30 by chamnit