gnea/grbl #184: random? errors

tcoleman978, 2017-05-01:
Grbl v1.1 on both Chinese clone uno and on (several) original new uno with 16U2 comm chip, also tried updated 16U2 drivers just in case. Drip fed with bCNC, UGS and others. From windows lap top and the final system Raspberry Pi. All do the same. With large programs like 40k+ lines of code, GRBL throws an error, usually 33 but I've also seen "no feed rate programmed" and others. I tried slowing down the feed rate to 100mm/min to no avail. What am I missing? Other people have made the systems work or it would be a big issue. Oh and the Gcode file is ok (several different files). I backplotted it in MasterCam at work and all is fine along with just sitting down and looking through the code.
Help!?
Thank You 
Todd

chamnit, 2017-05-01:
@tcoleman978 : Grbl is highly stable and reliable, so it's likely something with configuration. Can you provide your $$ settings AND your $I build info output? Thanks.

chamnit, 2017-05-01:
Also the error:33 is an invalid target. It most often has to do with improperly generated arcs. Grbl does some basic error checking of the g-code per LinuxCNC (generous) guidelines. This error is generated when: the target is the same as the current position in arc Radius mode; the incremental arc radius is off by 0.5mm or 0.1% radius (and 0.005mm). This error is also generated when a G38.x probe target is the same as the current position, but this is unlikely in your case.

There are reports that Fusion360 has this problem with Grbl when the arcs are generated by Fusion360 in inch mode. They go away when they are generated in mm mode. Other CAM don't have this problem, so I've suspected Fusion360. If you are using MasterCAM, I may need to investigate this a little further.

tcoleman978, 2017-05-01:
Ok, will have to retrieve the config and build info tonight when I get home from work but.... the device at work which is set up the same as at home and has displayed the same problem is as follows:
 Grbl 1.1f ['$' for help]
$0=10
$1=25
$2=0
$3=0
$4=0
$5=0
$6=0
$10=1
$11=0.010
$12=0.002
$13=0
$20=0
$21=0
$22=0
$23=0
$24=25.000
$25=500.000
$26=250
$27=1.000
$30=256
$31=0
$32=1
$100=160.000
$101=160.000
$102=250.000
$110=10000.000
$111=10000.000
$112=500.000
$120=800.000
$121=800.000
$122=10.000
$130=100000.000
$131=100000.000
$132=200.000
ok
ok
[VER:1.1f.20170131:]
[OPT:V,15,128]
ok
ok
default except for spindle max, laser mode and basic axis config 100, 110, 120, 130 for X and Y axes.
As for error 33 the math works out ok. The gcode was generated in LaserGRBL, and proofed on MasterCam, looking for bad G2/3 I-J's. This also happens on large files regardless what they were generated on. Sometimes reporting that there were multiple G codes on one line. This is why I thought it might be the non-16U2 usb comm problem that was reported.....no such luck.

chamnit, 2017-05-01:
@tcoleman978 : Nothing too out of the ordinary. Although you are getting pretty close to the max step rate Grbl can handle on the Arduino Uno and your max travel seems a bit high at 100meters. Can you supply the problem file? I'd like to see if I can reproduce the error.

tcoleman978, 2017-05-01:
with no home (yet) travels are set to "out of the way" dimensions.
how do I attach 43k lines of code?

chamnit, 2017-05-01:
Not sure if there is another way, but it is usually just providing a link. Dropbox works fine for this.

swarfer, 2017-05-01:
> There are reports that Fusion360 has this problem with Grbl when the arcs are generated by Fusion360 in inch mode. They go away when they are generated in mm mode. Other CAM don't have this problem, [...]

The problem is the postprocessor that comes supplied with Fusion and labelled for GRBL.  it does not work so well, and relies on a correctly set G28 and will output tool commands and other things GRBL does not support.  Like they never did read the GRBL wiki!

For the record, there is a much better one available from
https://github.com/Strooom/GRBL-Post-Processor/
which puts out an extra digit of precision for arcs, does not require G28, but does require a proper home position that is safe to use via G53.  It will assume G54 is the WCS is not set, or use the current setting if it is set (G54 to G59 only).

So, the extra digit.... in mm mode normal co-ords are 3 digits, arcs are 4 digits, in inches, 4 and 5 digits.
This is what I did in [SketchUcam](https://github.com/swarfer/sketchucam/releases) and it appears to be a total solution to the arc precision issues.

tcoleman978, 2017-05-01:
Ok, here is the offending program.
https://www.dropbox.com/s/ohsfn5790o1gm8p/ToddLine.nc?dl=0
let me know if you have any trouble as this is my first dropbox.
oh and dropping the max travels to 1000mm made no difference.

109JB, 2017-05-02:
I just ran your code through my interface program and got an error 33 on line 38071.

Here is line 38070 and 38071

G1 X39.496 Y0.045
G2 X39.202 Y0 I-0.293 J-0.045

So the arc starts at 39.496, 0.045
With the programmed I,J, the radius is  0.2432 and the center point is at 39.257, 0
the final point is only 0.055 from the center and is therefore not on the arc.

So, Grbl is correctly reporting an error because the final point is not on the arc using the i,j coordinates

chamnit, 2017-05-02:
@tcoleman978 : I can confirm that it fails at lines 38071-38073, starting on the arc on line 38071. I've ran it twice and fails at the same place. It's not random and primarily a problem with poor CAM software. For very small arcs like this one, the CAM really should be generating a line or a handful of them. G-code arcs are mathematically poorly posed and lead to errors like this. 

On Grbl's side, I could start checking for short arcs like this and ignore errors when under a certain size, but it isn't a standard error. Honestly though, I think you'd be better off by having your CAM generate G1's and no G2/3 arcs if you start running into these types problems. 

@109JB : Good to know about the txt file.

closed 2017-05-02 by chamnit

109JB, 2017-05-02:
As an aside, I have had no problems with the standard Fusion 360 post processor and all I ever use is inches. Not saying it isn't possible, but I have not encountered any arc errors from a F360 post. Also, it does not rely on G28, as G28 can be turned off when selecting post-processing options. As for the tool changes, even though grbl doesn't support M6T#, some GUI's do. Mine does and I want the M6T# commands in there. Editing the grbl post-processor file isn't that tough, and if using a gui that doesn't support it, you can simply add 4 lines to the post-processor file to create a post-processing option to not have tool changes. I just did it and those lines are:

in // user-defined properties
add this line: 
>useToolChange: true,       

or      
>useToolChange: false,

if you want false to be the default value

Then, around line 266, find this line:
>writeBlock("T" + toolFormat.format(tool.number), mFormat.format(6));

and replace with these 3 (actually just putting the above line in an if statement):

> if (properties.useToolChange) {
>        writeBlock("T" + toolFormat.format(tool.number), mFormat.format(6));
> }

tcoleman978, 2017-05-02:
Ok, seems that LaserGRBL is spewing garbage. I haven't had an run yet that works thru the whole output. Whats worse is the backplotters don't catch it and of course it happens near the end of a 1/2 to 3/4 hour run. Is there anything out there that is close to LaserGRBL that does work?
Thank a million for your help so far. Yea the drag and drop .txt IS a lot easier Thanx again..

109JB, 2017-05-02:
You could run it through Grbl in check mode.  Just send Grbl a $C command then stream the program. Grbl will go through it as fast as it can and check for errors like this. My experience is that check mode using my streaming program runs at about 9500 lines per minute so your program you linked would take less than 5 minutes in check mode.

fragmuffin, 2017-07-01:
@chamnit @109JB : what is the tolerance on G(2|3) arc verification?
I ask because I seem to have some shonky CAM output as well...

    G01 X10.75 Y-47.437
    G02 X8.271 Y-47.704 I-1.247 J-0.058 : 10.750,-47.437,-0.500 --> 8.271,-47.704,-0.500
    ; rad1=1.248348 ; from I & J parameters
    ; rad2=1.249602 ; from dX & dY
    ; error=0.001254 mm

Is 0.001254mm outside tolerance?

also, any recommendations for post-processing gcode files to smooth out issues like this?

swarfer, 2017-07-01:
The best solution I have found is to have the CAM output an extra digit of precision for the arc moves.
I added this to Stroooms postprocessor for Fusion360
https://github.com/Strooom/GRBL-Post-Processor
after proofing it in SketchUcam (CAM for Sketchup) which I maintain.

You don't say how you are generating your Gcode but maybe this will help you fix this.

fragmuffin, 2017-07-01:
Thanks @swarfer 
I'm using heekscad (heekscam) to generate gcode, but since this software seems to be abandoned, I should probably move to something else.
Thanks for your recommendation.

fragmuffin, 2017-07-04:
To answer my own question: "is 0.001254mm outside tolerance?"

[LinuxCNC documentation says **yes**](http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g2-g3)

> It is an error [when] the arc is projected on the selected plane, the distance from the current point to the center differs from the distance from the end point to the center by more than (.05 inch/.5 mm) OR ((.0005 inch/.005mm) AND .1% of radius)