gnea/grbl #185: cease to function when start cam job

mkeyno, 2017-05-02:
hi , I set the GRBL as laser mode and use GRBL plot software to test simple cutting test , but when I set feedrate and start the job it cease to function and only need hardware restart to respond again , can you tell me why it stopped in the line which is indicated in following image

firmware setting code as following
```
> $$ 
< $0=44 	(Step pulse time, microseconds)
< $1=49 	(Step idle delay, milliseconds)
< $2=27 	(Step pulse invert, mask)
< $3=5 	(Step direction invert, mask)
< $4=1 	(Invert step enable pin, boolean)
< $5=0 	(Invert limit pins, boolean)
< $6=0 	(Invert probe pin, boolean)
< $10=0 	(Status report options, mask)
< $11=0.000 	(Junction deviation, millimeters)
< $12=-2147483.648 	(Arc tolerance, millimeters)
< $13=1 	(Report in inches, boolean)
< $20=0 	(Soft limits enable, boolean)
< $21=0 	(Hard limits enable, boolean)
< $22=0 	(Homing cycle enable, boolean)
< $23=49 	(Homing direction invert, mask)
< $24=2147483.648 	(Homing locate feed rate, mm/min)
< $25=2147483.648 	(Homing search seek rate, mm/min)
< $26=250 	(Homing switch debounce delay, milliseconds)
< $27=0.000 	(Homing switch pull-off distance, millimeters)
< $30=0 	(Maximum spindle speed, RPM)
< $31=0 	(Minimum spindle speed, RPM)
< $32=1 	(Laser -mode enable, boolean)
< $100=-2147483.648 	(X -axis steps per millimeter)
< $101=0.000 	(Y -axis steps per millimeter)
< $102=0.000 	(Z -axis steps per millimeter)
< $110=0.000 	(X -axis maximum rate, mm/min)
< $111=0.000 	(Y -axis maximum rate, mm/min)
< $112=-0.000 	(Z -axis maximum rate, mm/min)
< $120=0.000 	(X -axis acceleration, mm/sec^2)
< $121=0.000 	(Y -axis acceleration, mm/sec^2)
< $122=-0.000 	(Z -axis acceleration, mm/sec^2)
< $130=2147483.648 	(X -axis maximum travel, millimeters)
< $131=-0.000 	(Y -axis maximum travel, millimeters)
< $132=-0.000 	(Z -axis maximum travel, millimeters)
< ok
```

chamnit, 2017-05-02:
@mkeyno : Have you even bothered to configure Grbl?! All of your settings are completely screwed up. Your steps/mm has one large negative value and the others are zero. No maximum rate or acceleration settings. Your arc tolerance and junction deviation values are way off from the default. The latter is important for the planner, which needs to be > 0.0 and around 0.01mm. Reset your settings with a `$RST=*` command and start over.

closed 2017-05-02 by chamnit

mkeyno, 2017-05-02:
@chamnit I'm so sorry for that , I've just noticed that , but I never expected such setting , because is not what I set before , I don't know why this setting have change by itself

swarfer, 2017-05-13:
We have seen this before!  This happens when the step rate is way over 30khz, no?
put your motor settings into this http://swarfer.co.za/cnc/grblcalc.html and see if it exceeds the 30khz step rate limit.