gnea/grbl #188: homihg issue (???)

DmitrSou, 2017-05-05:
Help please I have no ideas left what to check...
Everything the same for 0.9j and 1.1

After sending $H I expect moving to the endstops (Z, then XY). Instead of that I get moving Z twice -1mm, then moving XY (at the same time) twice -1mm. In console%
>>> $H
ok

The position shows
[verbose]<Idle|MPos:-1.000,-1.000,-1.000|Bf:15,128|FS:0,0>
[verbose]<Idle|MPos:-1.000,-1.000,-1.000|Bf:15,128|FS:0,0>
[verbose]<Idle|MPos:-1.000,-1.000,-1.000|Bf:15,128|FS:0,0|WCO:-10.634,-1.000,-1.000>
[verbose]<Idle|MPos:-1.000,-1.000,-1.000|Bf:15,128|FS:0,0|Ov:100,100,100>

no matter where actually spindel is located...

I expect the same behavior os for my 3D printer (Marlin) - moving Z until endstop (with bounce) then XY...

My configuration:

Arduino Uno + CNC Shield v3
Settings:
>>> $$
$0 = 10    (Step pulse time, microseconds)
$1 = 255    (Step idle delay, milliseconds)
$2 = 100    (Step pulse invert, mask)
$3 = 0    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 0    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 19    (Status report options, mask)
$11 = 0.010    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$20 = 0    (Soft limits enable, boolean)
$21 = 0    (Hard limits enable, boolean)
$22 = 1    (Homing cycle enable, boolean)
$23 = 0    (Homing direction invert, mask)
$24 = 25.000    (Homing locate feed rate, mm/min)
$25 = 500.000    (Homing search seek rate, mm/min)
$26 = 250    (Homing switch debounce delay, milliseconds)
$27 = 1.000    (Homing switch pull-off distance, millimeters)
$30 = 1000    (Maximum spindle speed, RPM)
$31 = 0    (Minimum spindle speed, RPM)
$32 = 0    (Laser-mode enable, boolean)
$100 = 314.925    (X-axis travel resolution, step/mm)
$101 = 314.925    (Y-axis travel resolution, step/mm)
$102 = 283.000    (Z-axis travel resolution, step/mm)
$110 = 1000.000    (X-axis maximum rate, mm/min)
$111 = 1000.000    (Y-axis maximum rate, mm/min)
$112 = 1000.000    (Z-axis maximum rate, mm/min)
$120 = 30.000    (X-axis acceleration, mm/sec^2)
$121 = 30.000    (Y-axis acceleration, mm/sec^2)
$122 = 10.000    (Z-axis acceleration, mm/sec^2)
$130 = 320.000    (X-axis maximum travel, millimeters)
$131 = 500.000    (Y-axis maximum travel, millimeters)
$132 = 40.000    (Z-axis maximum travel, millimeters)
ok

It's work fine if I type GCode in console. If "hard limit" is ON everything blocks when endstops is triggered.

In verbose mode I got following changes when trigger endstops manually

[verbose]<Idle|MPos:0.000,0.000,0.000|Bf:15,128|FS:0,0|Pn:X>
[verbose]<Idle|MPos:0.000,0.000,0.000|Bf:15,128|FS:0,0>
[verbose]<Idle|MPos:0.000,0.000,0.000|Bf:15,128|FS:0,0>
[verbose]<Idle|MPos:0.000,0.000,0.000|Bf:15,128|FS:0,0|WCO:-10.634,-1.000,-1.000>
[verbose]<Idle|MPos:0.000,0.000,0.000|Bf:15,128|FS:0,0|Ov:100,100,100>
[verbose]<Idle|MPos:0.000,0.000,0.000|Bf:15,128|FS:0,0>
[verbose]<Idle|MPos:0.000,0.000,0.000|Bf:15,128|FS:0,0>
[verbose]<Idle|MPos:0.000,0.000,0.000|Bf:15,128|FS:0,0>
[verbose]<Idle|MPos:0.000,0.000,0.000|Bf:15,128|FS:0,0>
[verbose]<Idle|MPos:0.000,0.000,0.000|Bf:15,128|FS:0,0>
[verbose]<Idle|MPos:0.000,0.000,0.000|Bf:15,128|FS:0,0|Pn:X>
[verbose]<Idle|MPos:0.000,0.000,0.000|Bf:15,128|FS:0,0>
[verbose]<Idle|MPos:0.000,0.000,0.000|Bf:15,128|FS:0,0>
[verbose]<Idle|MPos:0.000,0.000,0.000|Bf:15,128|FS:0,0|WCO:-10.634,-1.000,-1.000>
[verbose]<Idle|MPos:0.000,0.000,0.000|Bf:15,128|FS:0,0|Ov:100,100,100>
[verbose]<Idle|MPos:0.000,0.000,0.000|Bf:15,128|FS:0,0>
[verbose]<Idle|MPos:0.000,0.000,0.000|Bf:15,128|FS:0,0>
[verbose]<Idle|MPos:0.000,0.000,0.000|Bf:15,128|FS:0,0>
[verbose]<Idle|MPos:0.000,0.000,0.000|Bf:15,128|FS:0,0>
[verbose]<Idle|MPos:0.000,0.000,0.000|Bf:15,128|FS:0,0|Pn:Z>
[verbose]<Idle|MPos:0.000,0.000,0.000|Bf:15,128|FS:0,0|Pn:Z>
[verbose]<Idle|MPos:0.000,0.000,0.000|Bf:15,128|FS:0,0>
[verbose]<Idle|MPos:0.000,0.000,0.000|Bf:15,128|FS:0,0>
[verbose]<Idle|MPos:0.000,0.000,0.000|Bf:15,128|FS:0,0|WCO:-10.634,-1.000,-1.000>
[verbose]<Idle|MPos:0.000,0.000,0.000|Bf:15,128|FS:0,0|Ov:100,100,100>
[verbose]<Idle|MPos:0.000,0.000,0.000|Bf:15,128|FS:0,0|Pn:Z>
[verbose]<Idle|MPos:0.000,0.000,0.000|Bf:15,128|FS:0,0>
[verbose]<Idle|MPos:0.000,0.000,0.000|Bf:15,128|FS:0,0>

Any comment/advise would be helpful. What I have to check/configure?

vMeph, 2017-05-05:
The -1mm that moves is what you have set for homing switch pull off $27 = 1.000 (Homing switch pull-off distance, millimeters, when homing it always repeats 2 times the movment 1 in higher feedrate and second in lower feedrate to have it better precision

chamnit, 2017-05-05:
This also means that something is falsely triggering your limit switches right at the beginning of motion. This is almost always caused by EMF from your stepper motors and the lack of electrical interference filters/shielding. Add a low pass filter circuit as diagramed in the wiki. This should fix it.

closed 2017-05-05 by chamnit

DmitrSou, 2017-05-05:
"it always repeats 2 times the movment" - but it moves in the same direction. As for Z (it's shortest in my case) such action leads to stumble spindel in the table after several attempts...

"something is falsely triggering your limit switches" - sould it be reflected in verbose mode? If verbose is turn ON i clearly see triggering my limiters - no sign of EMF... Also behaviour is repeateble not random

My limiters is normally short to +5V When opened they pulled to GND by 1K resistors...

By the way - is my expectations about homing alrorythm right (mowing until endstop is triggered)?

Try to get oscilloscope and watch signals.

109JB, 2017-05-05:
Sounds like you need to set $5=1 to invert the limits.

chamnit, 2017-05-05:
@DmitrSou : 
- "repeats 2 times in the same direction" This is because Grbl pulls off once after it finds the switch, ad then pulls off again after slowly touching off on the switch again. Or bounce, as you say.

- "reflected in verbose mode" No. EMF and interference is high frequency and intermittent. Grbl polls the limit pins for their state when queried. 99% of the time, a poll will not pick up EMF. Nor will a voltmeter. You need an oscilloscope. Your stepper motors draw a lot of current when they start motion, this is usually the cause of EMF on the limit pins.

- You need a small capacitor to create an RC filter, aka low-pass filter, as the limit swithch wiring wiki page shows.

- "mowing into endstop" : This shouldn't happen if configured correctly. The pull-off motion should be away from the switch. If it's moving toward, then you have your homing search direction wrong or your coordinate frame wrong. Alter the homing direction mask (or step direction mask).

@109JB : No, his status reports show the switches as "not triggered" when idle and "triggered" when manually triggered. So they are configured correctly. This is a classic case of EMF and interference.

DmitrSou, 2017-05-06:
Looks like @chamnit is right. I've just received spare CNC Shield and run GRBL manually. Works perfect. On the next week will apply modification to my CNC and check. Thank you for pointing the way.

DmitrSou, 2017-05-10:
Now everything is perfect - I added capacitors as pointed in wiki. Thanks a lot.