gnea/grbl #189: Arduino is crashed when setting position by G-Code.

TsubokuLab, 2017-05-05:
I'm used Grbl v0.9 in my application and now updated to v1.1f.
but, my Arduino is crashed when setting position by G-Code.
I tried simple test using Arduino serial monitor.
Do I have any mistakes?

Left: sent messages.
Right: receive messages.(Arduino Serial Monitor)
[image]

When tested again, it received error with garbled characters.
[image]

This is version and config info. (from $i and $$ commands)
[image]

chamnit, 2017-05-05:
You are exceeding the maximum step rate supported by nearly 3x. Grbl can do up to 30kHz, as stated on the front page and the wiki. You are at about 84kHz (250step/mm * 20000 mm/min / 60 sec/min). This is limited by how fast the Arduino CPU is.

closed 2017-05-05 by chamnit

TsubokuLab, 2017-05-05:
Thank you so much.
You were right. and sorry for my poor question.