gnea/grbl #193: Problem GRBL test false!

RusSerg, 2017-05-12:
Im using arduino uno and GRBL 1.1

For test stepper motor im write.

```
$100=1
G01 X1000 F1
```

Im  wait 1 step per 1 second for stepper motor! Receive packet 4 stepper steps and pause and again 4 steps.

Test false!

Stepper motor rotate direction right.
Stepper driver work fine and change another with false test!

chamnit, 2017-05-12:
@RusSerg : Grbl is designed for real-world CNC applications. Not bench tests. So it has a lower limit of steps about 240 steps/sec by default. You can decrease this if by disabling ADAPTIVE_MULTI_AXIS_STEP_SMOOTHING in config.h. This will lower it to about 4 steps/sec. This is all limited by the 16-bit timer.

closed 2017-05-12 by chamnit

RusSerg, 2017-05-12:
1) Why GRBL not send warning message about slow speed? Im create pyroprinter with 1 stepper motor steps  1mm and im no need big speed.

 

> _You can decrease this if by disabling ADAPTIVE_MULTI_AXIS_STEP_SMOOTHING in config.h. This will lower it to about 4 steps/sec. This is all limited by the 16-bit timer._

2) Im try 4 steps

```
$100=4
G01 X1000 F1
```
And result packet steps and pause and again packet steps and pause e.t.c. 
# Problem!

chamnit, 2017-05-12:
@RusSerg : Sure. I can add a slow speed warning in future versions, but if you hadn't noticed, Grbl has completely maxed out the 328p processor on the Arduino. There is literally no room to add things like this. The remaining space is reserved for bug fixes only.

I think you need to read up on how g-code works. You are still commanding step rates way below what Grbl supports at 0.067 steps/sec. `F` feed rate is in terms of mm/_**minute**_.

RusSerg, 2017-05-12:
Im recompile with disabling ADAPTIVE_MULTI_AXIS_STEP_SMOOTHING in config.h

And test again.
Im try:

```
$100=1
G01 X1000 F4
```
Result no **pauses**
```
$100=1
G01 X1000 F1
```
Result 16 steps and **pause** and 16 steps and **pause** e.t.c.

Why GRBL have pauses?

chamnit, 2017-05-12:
It's likely a numerical round-off error in the code, but to me this is still a non-issue. Because you are still running at less than 1 step sec. 4 times less than what the Arduino is capable of doing. Again this is all due to the 16-bit timer and how high the prescalers go.

chamnit, 2017-05-12:
Correction. The lowest step rate Grbl supports with AMASS enabled is about 30 steps/second. Not 240.

With AMASS disabled, Grbl supports down to 4 steps/second. That's still the same.

RusSerg, 2017-05-12:
With AMASS disabled, Grbl work without pauses.
```
$100=4
G1 X1000 F60
```
In future slow speed write in Grbl code?

chamnit, 2017-05-12:
@RusSerg : Yes. Grbl just needs a better timer. The ARM version on the SAMD21 (coming very soon) has a 24 bit timer with configurable clocks. It'll be able to resolve just about any rate you will need.

langwadt, 2017-05-12:
if one really needed such slow step rates, it shouldn't be too hard to add a software prescaler on the timer interrupt

RusSerg, 2017-05-27:
Why problem timer? On slow speed may be use additional software timer?

chamnit, 2017-05-30:
@RusSerg : Feel free to implement a software pre scalar yourself. There is no room left on the 328p for Grbl's general release. Like I said before, this isn't a problem with the SAMD21 due to the higher resolution timer.