gnea/grbl #195: CMD_JOG_CANCEL leads to hang?

egradman, 2017-05-15:
I'm incorporating jogging into a frontend I've written.  I've followed the basic outline in the "Joystick Implementation," but sending CMD_JOB_CANCEL leads to a lockup.  Whenever my commanded jog d_pos goes to zero, I send 0x85 to GRBL as instructed.  The machine goes to the idle state.  It continues to give status updates, but it won't drive motors (jogging or otherwise) until I issue a ^x.

chamnit, 2017-05-15:
@egradman : This is a duplicate [issue](https://github.com/gnea/grbl/issues/95). Please give it a read. If you disagree, post a new issue with reasons why.

closed 2017-05-15 by chamnit

egradman, 2017-05-15:
HI @chamnit.  I've looked through that issue, and from what I understand it deals with a situation where CMD_JOG_CANCEL doesn't abort the movement.  In my case, the movement is being aborted just fine!  It just leaves grbl in a state where it won't accept movement commands.

I do see a comment about waiting until the idle state is reached before issuing additional CMD_JOG_CANCEL commands, but I wonder if that refers to an older version of the codebase... looking at `serial.c` it looks like CMD_JOG_CANCEL is a no-op if I'm not in the jogging state.

I'm sorry if I've misunderstood the other issue.

Also, if it hasn't been said enough here... this software is amazing!

109JB, 2017-05-16:
I too experienced occasional problems with jog cancellation.  I previously had my GUI set up to send numerous jog commands of a small distance repeatedly and then send the jog cancel to stop things. I experienced problems with this method and ultimately changed my approach which has worked flawlessly. Here is what I did:

1. On a key-down or mouse-down event my GUI sends a single jog command . If the soft limits are enabled then the GUI has to calculate the distance from the current position to the machine limit to avoid a "error: 15".  If soft limits aren't enabled then it just sends a single jog command with a huge distance.  In either case, it send only one jog command regardless of how long the key is held down.

2. On the key-up or mouse-up event, the GUI sends the 0x85 jog cancellation command. 

Doing it the way described above has resulted in flawless performance.

egradman, 2017-05-16:
Yes!  Your old approach is exactly the approach I've been taking.  It didn't even occur to me to perform long jogs.  I'll try this now.  Thanks.

Eric Gradman
Mad Inventor & CTO

P: 213.268.6253 | W: twobitcircus.com

TWO BIT CIRCUS
Engineering Entertainment
Inventors | Developers | Performers