gnea/grbl #196: XON/XOFF flow control caveats not in v1.1 WIKI

109JB, 2017-05-18:
I noticed that the new V1.1 WIKI does not include the 4 paragraphs description and cautions regarding XON/XOFF flow control that the v0.9 WIKI had.  Is this because the XON/XOFF is now "fixed", or do the issues and admonitions from v0.9 still apply.  If they still apply, then I believe the section from the V0.9 WIKI should be copied over to the V1.1 WIKI, which I can do if applicable.

The reason I brought this up is because while searching for some stuff I came across a post in another forum that basically said you can now use XON/XOFF with Grbl and it referenced the section of the code in serial.c, but I see that the config.h still has the comment warning that XON/XOFF might now work right.

langwadt, 2017-05-18:
there is not longer any xon/xoff code in serial.c, so I would assume no longer supported

109JB, 2017-05-18:
Thanks.  I see that the code has been removed and at some time the define in config.h was removed.  Curious that the config.h that I have does still have it, but I have been using a forked version.  I should have checked the current build before commenting.  In any case, I think I will add a few sentences to the WIKI to let people know that XON/XOFF has been removed completely.

closed 2017-05-18 by 109JB