gnea/grbl #197: G7 Code feature for Raster image (CNC Laser)

mkeyno, 2017-05-21:
Before I came back again to GRBL I use modified version of [Marlin firmware ](https://github.com/TurnkeyTyranny/buildlog-lasercutter-marlin/blob/master/Marlin/Marlin_main.cpp#L1070) combine the extension for[ inskscape](https://github.com/TurnkeyTyranny/laser-gcode-exporter-inkscape-plugin/blob/master/turnkeylaser.py#L90) to convert image to raster and feed it to my 100W CNC laser for engraving , it shows very high speed for engraving but Marlin firmware has not feature such as real time overriding code to control machine over the G-code, so I return to new GRBL release firmware with laser mode feature , but thanks to @arkypita & @svenhb ,after  couple of try with different G-code generator such as [LaserGRBL ](https://github.com/arkypita/LaserGRBL)& [GRBL-Plotter](https://github.com/svenhb/GRBL-Plotter)  and different GRBL setting I've noticed Sending a bitmap/raster image in G-Code syntax (which is designed for vector instructions) is a terrible waste of bytes due to hardware limit to current implementation on Arduino uno , it is pretty low band wide buffer for such task , according to @mayhem2408 note, "The G7 code used in the Marlin format is very effecient, but my experience with it has shown it is much more computationally intensive. The maximum feed I can get out of Marlin using G7s base64 is around F7000 before it starts to stutter, and it's not because of bandwidth. G7s base64 should be able to transfer just over 7000 pixels per second, which at 10 pixels per mm, can in theory hit a feed of over F40000. The other problem as far as GRBL is concerned, there is no room left in the Arduino Unos memory to add such a massive feature. It is something that may be added to the GRBL-Mega version as it has more room" ,[ this is discuss page.](https://github.com/arkypita/LaserGRBL/issues/38#issuecomment-302833773) 
I know this wont be done on Arduino and thankfully  couple of GRBL version run on other board such STM32F103, I was wondering if official developer of this community will have plan to implement such nice feature on such STM board , thanks to [Roger Clarck, his repo ](https://github.com/rogerclarkmelbourne/Arduino_STM32)is real mature and robust and [STM32F103C8 is ](http://www.rogerclark.net/stm32f103-and-maple-maple-mini-with-arduino-1-5-x-ide/)cheap as Arduino uno with much more powerful, also couple of GRBL enthusiast such as @langwadt, @Meekdai, @MaxwellXyao port it to STM hardware but not officially 

[image]

implementation of G7 on Marlin

python function to create raster code
```python
 def generate_raster_gcode(self, curve, laserPower, altfeed=None):
...
```
code  for  G7 interpreter 
```cpp

 #ifdef LASER_RASTER
    case 7: //G7 Execute raster line
      if (code_seen('L')) laser.raster_raw_length = int(code_value());
	  if (code_seen('$')) {
		laser.raster_direction = (bool)code_value();
		destination[Y_AXIS] = current_position[Y_AXIS] + (laser.raster_mm_per_pulse * laser.raster_aspect_ratio); // increment Y axis
	  }
      if (code_seen('D')) laser.raster_num_pixels = base64_decode(laser.raster_data, &cmdbuffer[bufindr][strchr_pointer - cmdbuffer[bufindr] + 1], laser.raster_raw_length);
	  if (!laser.raster_direction) {
	    destination[X_AXIS] = current_position[X_AXIS] - (laser.raster_mm_per_pulse * laser.raster_num_pixels);
	    if (laser.diagnostics) {
          SERIAL_ECHO_START;
          SERIAL_ECHOLN("Negative Raster Line");
        }
	  } else {
	    destination[X_AXIS] = current_position[X_AXIS] + (laser.raster_mm_per_pulse * laser.raster_num_pixels);
	    if (laser.diagnostics) {
          SERIAL_ECHO_START;
          SERIAL_ECHOLN("Positive Raster Line");
        }
	  }
	  
	  laser.ppm = 1 / laser.raster_mm_per_pulse; //number of pulses per millimetre
	  laser.duration = (1000000 / ( feedrate / 60)) / laser.ppm; // (1 second in microseconds / (time to move 1mm in microseconds)) / (pulses per mm) = Duration of pulse, taking into account feedrate as speed and ppm
	  
	  laser.mode = RASTER;
	  laser.status = LASER_ON;
	  laser.fired = RASTER;
	  prepare_move();

      break;
#endif // LASER_RASTER
```

sample raster code by inkscape 

```
`; Raster data will always precede vector data
; Default Cut Feedrate 300 mm per minute
; Default Move Feedrate 2000 mm per minute
; Default Laser Intensity 10 percent
G28 XY; home X and Y

M5 ;turn the laser off

;(_)
;(** Layer: 10 [ppm=30,feed=2500] ****)
;(** Laser Power: 10 ****)
;(** Feed Rate: 2500.0 ****)
;(** Pulse Rate: 30.0 ****)
;(_********)
;(MSG,Starting layer '10 [ppm=30,feed=2500]')

;Beginning of Raster Image image4556 pixel size: 753x750
M649 S10 B2 D0 R0.09406
G0 X11.518 Y15.927 F2500

G7 $1 L68 DAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
G7 L68 DAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
G7 L68 DAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
G7 L68 DAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
G7 L68 DAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
G7 L68 DAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
G7 L68 DAAAAAAAAAAAAAAAAAAAAAAMHCiA3Tl9xg42XoqOlp6CZkoZ5bFpINSQTAQEAAAAAAAAA
G7 L68 DAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAABAhMkNUda
G7 L28 DbHmGkpmgp6WjopiNg3FfTjchCwcD

G7 $0 L68 DBQoQMVN1kKvG1eT09vn78efcybajiGxQNx0DAgEAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
G7 L68 DAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAQIDHTdRbIikt8rd59H89fb05NXGq5B2
G7 L56 DUzEQCgUAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA=

G7 $1 L68 DAQMFFCQ0MzEwIBAAAAAAAAAAAAAAAAAAAAAAAAAAAAAACRIcMERYcImitMbY4u339fv8
G7 L68 D9u9o287Br5yKdV9KOSkYEAgAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
G7 L68 DAAAAAAAAAAAACBAYKTpKX3SJnK9Bztvn7vX899n37ePYxrSiiXFYRDAcEgkAAAAAAAAA
G7 L36 DAAAAAAAAAAAAAAAAAAAAECAwMjM0JRUFAwE=

G7 $0 L68 DAwcMK0tqZ2ViQSAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAEiU5W39iscHQ2eLr8fb79P39
G7 L68 D9vbz7ebg183EtKOTc1IyIRAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
G7 L68 DAAAAAAAAAAAAECExUnOTo7TEzdfg5u3z99r99fz79vDr4tnQwbGif1s5JRIAAAAAAAAA
G7 L36 DAAAAAAAAAAAAAAAAAAAAIUJiZGdqSioLBwM=

G7 $1 L68 DBQsRP29gnJiUYzIBAAAAAAAAAAAAAAAAAAAAAAAAAAAAHDlWh7rt8vj9999999999999
G7 L68 D99999999999989fcrHtLMhkAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
G7 L68 DAAAAAAAAAAAAGDJLe6zb6PP99999999999999999999999999PLsuodWORwAAAAAAAAA
G7 L36 DAAAAAAAAAAAAAAAAAAAAMWOTmJyfcUESCwUA

`

When I compare this to the included g-code for, say the wolf.gcode, I see something much more reasonable:

`G28 ; home all

M5

;(_)
;(** LAYER: 10 ****)
;(_**)
;(MSG,Starting layer '10')

G00 X3.299782 Y118.552030
T01 (select tool)
M6 (tool change)

M5

G00 X3.299782 Y118.552030 F1400
M3
G02 S10.00 X3.548178 Y118.501119 I0.000000 J-0.631423 F1400
G02 S10.00 X3.676074 Y118.391017 I-0.110031 J-0.257148 F1400
G03 S10.00 X3.805945 Y118.305855 I0.156134 J0.096495 F1400
G03 S10.00 X4.136655 Y118.324427 I0.119469 J0.826362 F1400
G03 S10.00 X4.290573 Y118.498593 I-0.052748 J0.201711 F1400
G03 S10.00 X4.208141 Y118.860798 I-0.527850 J0.070353 F1400
G02 S10.00 X3.894311 Y120.088890 I1.636677 J1.072386 F1400
G02 S10.00 X4.381019 Y121.026364 I1.354478 J-0.108126 F1400
G03 S10.00 X4.518115 Y121.254128 I-0.257294 J0.310012 F1400
G03 S10.00 X4.521857 Y121.701707 I-1.117987 J0.233151 F1400
G01 S10.00 X4.429717 Y122.183697 F1400
G01 S10.00 X4.895852 Y121.878273 F1400
G03 S10.00 X5.715594 Y121.570679 I1.094221 J1.670000 F1400
G03 S10.00 X5.907575 Y121.737854 I0.023203 J0.167175 F1400
G02 S10.00 X5.950319 Y121.885151 I0.275167 J0.000000 F1400
```

mayhem2408, 2017-05-21:
@mkeyno This implementation of G7 is one of the first. It has evolved since then. One issue with this old G7 what how it calculated the change in direction. A newer implementation, and the one I have been working with for over a year, sets the starts on a line using G0/1 and then G7 to fill the line. This makes it much easier to work with. This is an example of the G7 code I use. You will notice the G0 before each G7 line to set the start position. It makes alignment easier because you don't have to extend the beginning or end of lines with useless '0' to align each line. The 'A' in the lines before are not for alignment, but room for acceleration.

> G0 X69.700 Y45.700 F1500.0; Line n.457
M649 S100 B2 D0 R0.1
G7 $0 L48 DAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAP///wA=
G7 L48 DAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA=
G7 L48 DAAAAAAAAAAAAAAAAAAAAAAAAAP///wAAAAAAAAAAAAAAAAA=
G7 L28 DAAAAAAAAAAAAAAAAAAAAAAAAAA==
G0 X57.300 Y45.800 F1495.0; Line n.458
M649 S100 B2 D0 R0.1
G7 $1 L60 DAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAP///wAAAAAAAAAAAAA=
G7 L60 DAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA=
G7 L48 DAP///wAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
G0 X69.700 Y45.900 F1501.0; Line n.459
M649 S100 B2 D0 R0.1
G7 $0 L56 DAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAP///wAAAAAAAAAA
G7 L56 DAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
G7 L56 DAAAAAAD///8AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA==
G0 X57.300 Y46.000 F1495.0; Line n.460
M649 S100 B2 D0 R0.1
G7 $1 L60 DAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAP///wAAAAAAAAAAAA==
G7 L60 DAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA==
G7 L52 DAAAA////AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA=

mayhem2408, 2017-05-21:
One other important reason for manually setting the start position on each line is it allows the X and Y axis to have difference resolutions. This comes in handy with dithered engraving. (i.e. on acrylic) Using R0.05 means 20 pixels per mm, but still only increasing Y by 0.1, effectively doubling the resolution, but without increasing the engraving time. It works really well.

```
M649 S100 B2 D0 R0.05
G0 X64.600 Y45.600 F1496.0; Line n.456
G7 $1 L64 DAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAP8AAAAAAAAAAAAAAAAAAAAA
G7 L24 DAAAAAAAAAAAAAAAAAAAAAA==
G0 X68.100 Y45.700 F1500.0; Line n.457
G7 $0 L52 DAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAP////////8=
G7 L44 DAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA=
G0 X62.300 Y45.700 F1499.0; Line n.457
G7 $0 L56 DAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAP////////8AAA==
G7 L40 DAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
G0 X58.800 Y45.800 F1502.0; Line n.458
G7 $1 L48 DAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAP////8=
G7 L48 D////AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA=

```

svenhb, 2017-05-21:
Hello John,

thanks for the interesting feature.

I checked G7 at LinuxCNC docu. and it has originally a completly different meaning: http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g7

So I fear GRBL will not implement the function you decribed here - too bad.

Best regards

Sven

 

Gesendet: Sonntag, 21. Mai 2017 um 18:42 Uhr

usbcnc, 2017-05-21:
@mkeyno I have ported 1.1 to STM32F103 chip. https://github.com/usbcnc/grbl. Check this
https://www.youtube.com/watch?v=5DBtbAYmfEE&t=31s
I you need help please contact me,

gerritv, 2017-05-21:
It could be implemented using an unused G code. Would have been wise of Marlin devs to have done that, the existing uses are well known.

chamnit, 2017-05-21:
I have looked at this in the past. I'm not fond of the G7 definition, but that doesn't mean there isn't a need for something like this. I'm still looking into what would work well. And will likely define this as a canned cycle, rather than G7.

closed 2017-05-21 by chamnit

mayhem2408, 2017-05-21:
@gerritv @chamnit It would take less that 2 minutes to change Marlin to use a different G code, like G99 and the change the inkscape plugin to reflect the same change. The actual G number is not important. I agree it should not be G7. I say G99 because it is not referenced by LinuxCNC or RepRap. I have seen a custom macro in UCCNC controller that uses M10027 as a similar command.

mkeyno, 2017-05-22:
@chamnit can you elaborate your plan ? , because as it mentioned , probably not can added to official GRBL Arduino nano, can you implemented to other official platform such as what @usbcnc done on STM32 ? it would be nice the community add such special feature to its official platform

chamnit, 2017-05-22:
@mkeyno : There have been lengthy discussions on this between several firmware devs, including myself [here](https://github.com/LaserWeb/LaserWeb4/issues/244#issuecomment-291571546). In short, there is no concensus. And there are other factors that affect performance that is not just gcode. In fact, it is only a very small part of the problem. 

Just to be absolutely clear on this. I will not elaborate any further since the thread already does that, and I will not work on specific ports of Grbl v1.1. All work will be done in the new, soon to be released hardware abstracted version of Grbl, so if @usbcnc chooses to, he will be able to pull in downstream changes easily for his ST platform. This way any work done in the HAL can be propagated easily to other ports that use it. Not just the ST, LPC, or SAMD21 versions.

mkeyno, 2017-05-23:
thanks @chamnit but those related issues not concluded to the meaningful way or code for such purpose , however as I intended to always sync to original (official )repo , how do you suggest to keep this synchronize ?
@mayhem2408 you  are so  familiar with this feature and @usbcnc  currently port original GRBL on the STM32 , is there a way so you guys could implemented such feature together ?

chamnit, 2017-05-23:
@mkeyno : Apologies, I linked the wrong issues thread. [This](https://github.com/LaserWeb/LaserWeb4/issues/99) is the one you want to read. Focus on the part where comm bandwidth isn't the limiting factor for Grbl. (Grbl on a 328 runs almost as fast as Smoothie on a Cortex M3, while the Grbl-LPC port runs the fastest of all of them in the discussion.) So, using G7 will not significantly improve raster jobs. 

Like I said before, gcode is only a very small part of the problem. I have a solution, which I will not publicly discuss until its closer to being released. And I will not elaborate further on it.

mayhem2408, 2017-05-23:
@chamnit As I have done heavy work on inkscape plugins for raster engraving in the past, once you have your solution in place, I'll happily test it and write whatever changes are needed into the plugin.