gnea/grbl #198: Jogging problem

cody82, 2017-05-23:
I made a jog GUI with buttons that send jog commands to Grbl.
When I press the button a jog command is sent.
When I release the button a jog cancel button is sent.

It works in most cases. But when I double click the button the last job command is not canceled and the machine continues to drive.
Also after doing this a few times Grbl ends up in Idle state and does not accept commands anymore.

Here is a log:
GrblReader Received: <Idle|MPos:0.000,0.000,6.684|FS:0,0>
GrblReader Received: <Idle|MPos:0.000,0.000,6.684|FS:0,0>
Jog start.
GrblReader Received: ok
Jog cancel.
GrblReader Received: <Jog|MPos:0.000,0.000,6.732|FS:48,0>
Jog start.
GrblReader Received: <Jog|MPos:0.000,0.000,6.804|FS:0,0>
Jog cancel.
GrblReader Received: ok
GrblReader Received: <Jog|MPos:0.000,0.000,6.836|FS:78,0>
GrblReader Received: <Jog|MPos:0.000,0.000,7.008|FS:100,0>
GrblReader Received: <Jog|MPos:0.000,0.000,7.216|FS:100,0>

cody82, 2017-05-23:
I solved the problem by waiting for the "ok" before canceling the jog.

closed 2017-05-23 by cody82

X3msnake, 2017-05-29:
Cool can you share the make?

cody82, 2017-05-31:
Do you mean the source code? Here it is: https://github.com/cody82/grblgui

X3msnake, 2017-05-31:
Yes Thanks :)