gnea/grbl #199: M03 not working outside of job

Bricbasher, 2017-05-24:
I have a quick question.  I am trying to use the command M03 S3 to turn on my laser to position it and focus it.  When I use that command nothing happens.  I can even use M03 S255 and nothing happens still.  However, when I run a laser job the laser comes on and engraves just fine.  I just need to be able to use the lower power command outside of a job in order to focus the laser.  I have been using UGS and Lasergrbl.

Thanks!

chamnit, 2017-05-24:
@Bricbasher : Make sure you are in a motion mode that turns the laser on, as [this wiki page](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Laser-Mode) says.

closed 2017-05-24 by chamnit

Bricbasher, 2017-05-24:
Just to clarify would I have to type a command like G1 M3 S3?