gnea/grbl #201: homing limit switch not recognized

Bair22, 2017-05-29:
Hi,
I just startet to set up a cnc-milling machine with grbl 1.1f
But homing doesn't work.
If I send '?' grbl recognizes the pushed limit switches. But if I run homing cycle the axis won't stop when limit switch is pushed. 

Can somebody help me?

My config:
Grbl 1.1f ['$' for help]
[MSG:'$H'|'$X' to unlock]
$$
$0=10
$1=25
$2=0
$3=7
$4=0
$5=0
$6=0
$10=3
$11=0.010
$12=0.002
$13=0
$20=0
$21=1
$22=1
$23=4
$24=25.000
$25=100.000
$26=250
$27=5.000
$30=1000
$31=0
$32=0
$100=250.000
$101=250.000
$102=250.000
$110=500.000
$111=500.000
$112=500.000
$120=10.000
$121=10.000
$122=10.000
$130=200.000
$131=200.000
$132=200.000

Thanks

X3msnake, 2017-05-29:
Are you wiring the shitches as NC or NO?

Bair22, 2017-05-29:
they are NO switches

X3msnake, 2017-05-29:
I presume you already read trough this correct?

https://github.com/gnea/grbl/wiki/Wiring-Limit-Switches

Bair22, 2017-05-29:
correct. with grbl 0.9i homing is working.

X3msnake, 2017-05-29:
You were using 0.9 and the homing was working?

Bair22, 2017-05-29:
Yes. And then I updated to 1.1 and no it doesn't recognize the pushed limit switches...

X3msnake, 2017-05-29:
What is the reported status of the limit switches when you do the debugging?

https://discuss.inventables.com/t/guide-using-grbl-to-debug-your-homing-switches/16588

Bair22, 2017-05-29:
With  `?` ?

When I push the limit switch of z-axis it says `<Alarm|MPos:0.000,0.000,0.000|Bf:15,127|FS:0,0|Pn:Z|WCO:0.000,0.000,0.000>`
with X and Y  	respectively  Pn:X or Y

Bair22, 2017-05-29:
Only Problem I have with 0.9 afer homing I run into Alarm-Mode...
Don't know what I'm doing wrong

X3msnake, 2017-05-29:
can you post a picture of your wiring?

Bair22, 2017-05-29:
I dont have a picture. But its wired like in the schematic.
[image]

swarfer, 2017-05-29:
> Only Problem I have with 0.9 afer homing I run into Alarm-Mode...
Don't know what I'm doing wrong

This usually means that the homing pull off distance, $27, is too small.
OR
you have interference in the wires.  The wiki tells you how to add suppression.
https://github.com/gnea/grbl/wiki/Wiring-Limit-Switches

Bair22, 2017-05-29:
what is typicall for $27? I have $27=5.000

chamnit, 2017-05-29:
@Bair22 : Add an RC filter. You have interference in your wires. If you see an alarm or error code, it would help if you told us what it was. My guess is that you have the homing directions reversed, the interference is causing the homing cycle to skip the search cycle and go right into the pull-off motion. You should be observing the homing cycle moving about 5mm then failing with an `ALARM:8` code.

closed 2017-05-29 by chamnit