gnea/grbl #202: enhancement: support for 'GOTO' statement

42loop, 2017-05-30:
for use cases with very simple motion commands, it would be great to avoid back-to-back mounting of a gcode sender to grbl by supporting the goto statement in the startup blocks

109JB, 2017-05-30:
I for one have no idea what you are talking about. Please explain further

X3msnake, 2017-05-30:
This is what he is speaking

http://www.cnccookbook.com/CCCNCGCodeIFGOTOWHEN.htm

I'm preatty sure think this is a better job for the Sender other then GRBL

chamnit, 2017-05-30:
IMHO, giving users that ability is quite dangerous. GOTO ignores any modal states set in between. Regardless, this is not a Grbl issue. It's a GUI issue, since Grbl does not care how g-code is sent to it.

closed 2017-05-30 by chamnit