gnea/grbl #204: Set direction pins directly after obtaining new segment

Schildkroet, 2017-06-04:
Hi,
i'm not sure if this problem will be still relevant with the new HAL, but i still want to address it.

In line 313 in stepper.c:
`// Set the direction pins a couple of nanoseconds before we step the steppers`
`DIRECTION_PORT = (DIRECTION_PORT & ~DIRECTION_MASK) | (st.dir_outbits & DIRECTION_MASK);`

The direction pins are set just a couple of ns before the steppers (with higher clock on ARMs this will be even less time), but some stepper drivers require the signals to be stable for a certain time before a step (e.g. 5us). This might cause 1 step in the wrong direction.

So it would be best to update the direction pins directly after obtaining the new segment in line 355:
`st.dir_outbits = st.exec_block->direction_bits ^ dir_port_invert_mask;`
`// Update direction...`

mfg

langwadt, 2017-06-04:
it will works and saves a few cycles,  it won't work the define STEP_PULSE_DELAY  though that could probably be removed anyway

chamnit, 2017-06-04:
The entire stepper driver has been abstracted out. It's up to the port to implement it best for the processor it's on.

closed 2017-06-04 by chamnit