gnea/grbl #205: status report after hard/softlimit

langwadt, 2017-06-05:
is it intentional or just a lack of program space the reason for not responding to "?" after hitting a limit?  
The comments indicates that at one point it was supposed to

chamnit, 2017-06-07:
Unintentional omission. I meant to add it, but forgot before the v1.1 release. I can't make any changes to the interface at this point, but will address it in the next major release.

closed 2017-06-07 by chamnit

langwadt, 2017-06-07:
I see you point, though it wouldn't really change the interface, would just add the option to always poll 
for status. And to be really useful there would probably have to be more states, or it'll still be impossible to tell if ALARM is from reset or error

chamnit, 2017-06-07:
Yes I understand that it won't change the interface that much, but I always here complaints whenever small tweaks like this occur, especially from OEMs that have their own GUI software. They don't want to spend time to constantly make small updates to their code like this and go through another testing cycle. That can be be draining and costly. Either way, this, along with status reports during a homing cycle, will be add in the next version.