gnea/grbl #206: step/mm in corexy system

adibiut, 2017-06-06:
Hi everybody.
As you know, in CoreXY mechansim for moving along x axis, two motors should turn together.
so, how should I calculate step/mm? ($100 and $101 in v1.1f config)

mayhem2408, 2017-06-06:
@adibiut CoreXY geometry is different to traditional Cartesian, but the steps are the same. If a Cartesian is 80 steps per mm, so it a CoreXY. You need to know how many steps per revolution your steppers are, how many micro steps your controllers are, your belt tooth pitch and the number of teeth on the stepper pulley. Let me know these 4 things and a simple calculation will work it out for you.

adibiut, 2017-06-06:
Dear @mayhem2408 
thanks for your reply.
I know the calculations for Cartesian system,but according to the following formula,for moving toward each axis,two motors should work together.
Assume that each motor is 200steps/rev, microstep is 1/8, and the diameter of pulley is 16mm (pitch*teeth). 
So how do you define steps/mm for x and y axis?
[image]

mayhem2408, 2017-06-06:
It is exactly the same formula. (200 * 8 ) / (pitch * #Teeth)
Pitch * Teeth does not equal diameter of pulley, but the diameter is not needed in the formula anyway. You need the circumference in mm.
My CoreXY is (200 Steps * 16 microsteps) / (2mm Pitch * 16 teeth) = (200 * 16) / (2 * 16) = 3200 / 32 = 100 Steps per mm. For CoreXY, X and Y must be same.

adibiut, 2017-06-06:
@mayhem2408  Thank you.solved

closed 2017-06-06 by adibiut

Fusseldieb, 2020-08-12:
Came here facing a very similar question. After searching a lot, nobody seems to understand the question.

The thing is, **CoreXY systems must be calibrated differently**! If you print a parallel (to the bed) cube and then try to measure the X and Y side of the cube and tune the X and Y parameter, your results will worsen every adjustment. 

In my case:

```
First adjustment:
x: 29.5
y: 30.22

Second:
x: 29.31
y: 30.73

Third (This time I even saw that it was stretched out)
x: 28.62
y: 31.52
```

If you need to adjust your X and Y in your CoreXY system, print the cube in a exact **45° angle**. While printing, make sure that when it's printing one of the sides of the cube that one motor is _moving_, while the other is standing _perfectly still_, and _vice-versa_.

With the print done, measure X and Y of the cube as you would normally and these are your true X/Y axis which you can then calibrate in your `configuration.h` or using the command [M92](https://marlinfw.org/docs/gcode/M092.html) and then M500 to save. Since it's rotated 45° I (we?) don't know which is X and Y, but playing around with the calibration you quickly guess it.