gnea/grbl #207: Not working on mini pro 8Mhz

akiAbtahi, 2017-06-07:
I was trying to run grbl on a mini pro running at 8 MHz (3.3V). The response is strange, for instance it requires two CR after $ to respond. For $$ it responds with error 3 ...... , responds to ? correctly. Anybody have any ideas on how to resolve this?
Thanks

JBSchueler, 2017-06-07:
8Mhz will decrease performance but the uart communication should be ok.
What kind of uart chip does it have (e.g. FTDI)?

akiAbtahi, 2017-06-07:
It's a FTDI. You're right it's probably a problem with serial comms. Other sketches with lower baud work fine so I should try to reduce the baud rate in grbl. I'll try this evening,
Thank You!

chamnit, 2017-06-07:
8Mhz builds are not supported, nor have been tested. Theoretically it should work. If you are using the Makefile, you'll need to edit it to denote 8 MHz.

closed 2017-06-07 by chamnit