gnea/grbl #208: Qestion About G code sender for this 4 axis grbl fork

Mareckil, 2017-06-07:
What G code sender are you using?

109JB, 2017-06-07:
You posted this in the main Grbl v1.1 Issues section. This isn't a 4th axis fork, and since this repository is for UNO powered Grbl installs there probably won't be. You should post in the issues area for the specific fork you are referring to since there could be coding differences that make the 4th axis implementation different.

As for me, I use my own GUI for sending to the 4th axis fork by @dguerizec that is at this link:
https://github.com/dguerizec/grbl-Mega-4axis
My GUI also works with the 4th axis by @LETARTARE at the link below if the 4th axis is named A:
https://github.com/LETARTARE/grblQ-Mega-4axes

I plan to release the new version of my GUI once I have some other things worked out in it.  Just have to find the time to get it done.

closed 2017-06-07 by chamnit

winder, 2017-06-07:
@109JB what special features do you have to make your GUI "4th axis enabled"? Do you go beyond looking for an extra set of coordinates in the status messages?

109JB, 2017-06-07:
@winder My GUI has full implementation for several canned cycles which requires it to interpret the G-code  from the user and then interpret it and change the code sent into what Grbl understands.  For example, a G83 peck drilling cycle requires "R" "L" and "Q" words and the GUI also has to know if G98 or G99 is active.  All of this is not native to Grbl so the GUI handles it and changes a single G83 line into a bunch of G0 and G1 commands it then sends to Grbl.  Because of this, the GUI has its own G-code parser, error checking, etc.

I have been doing some re-writing of my GUI code for this and other things which is what is holding up my releasing it. I am adding probing routines, user macros, had to change how it reads and stores Grbl settings due to extra 4th axis parameters, etc.

As for 4th axis implementation,  I changed the parser for status responses and re-wrote it in such a way that any number of axes can be supported. Then in my G-code parser, I had to allow for the "A" word to not raise an error within the GUI.  There will be more added for the 4th axis implementation in mine, but that was all I had to do to get it working for now.

winder, 2017-06-07:
@109JB I'm looking forward to inspecting the code if you release it. Sounds like you have some well thought out features. After several component rewrites, I've moved a modular approach for things like the gcode parser and visualizer in UGS, it's a fun challenge!