gnea/grbl #209: GRBL 1.1 slow jogging

clarkdv, 2017-06-07:
I'm running v1.1f and moving the spindle manually is now excruciatingly slow either by keyboard or clicking the buttons in GRBL Platform. Changing Step Size or Feed Rate speeds it up slightly but it's still way slower than it used to be with 0.9. Any ideas?

My GRBL Settings:

$0 = 10 (Step pulse time, microseconds)
$1 = 255 (Step idle delay, milliseconds)
$2 = 0 (Step pulse invert, mask)
$3 = 0 (Step direction invert, mask)
$4 = 0 (Invert step enable pin, boolean)
$5 = 0 (Invert limit pins, boolean)
$6 = 0 (Invert probe pin, boolean)
$10 = 3 (Status report options, mask)
$11 = 0.010 (Junction deviation, millimeters)
$12 = 0.002 (Arc tolerance, millimeters)
$13 = 0 (Report in inches, boolean)
$20 = 0 (Soft limits enable, boolean)
$21 = 0 (Hard limits enable, boolean)
$22 = 1 (Homing cycle enable, boolean)
$23 = 0 (Homing direction invert, mask)
$24 = 25.000 (Homing locate feed rate, mm/min)
$25 = 600.000 (Homing search seek rate, mm/min)
$26 = 244 (Homing switch debounce delay, milliseconds)
$27 = 4.000 (Homing switch pull-off distance, millimeters)
$30 = 1000 (Maximum spindle speed, RPM)
$31 = 0 (Minimum spindle speed, RPM)
$32 = 0 (Laser-mode enable, boolean)
$100 = 40.000 (X-axis travel resolution, step/mm)
$101 = 40.000 (Y-axis travel resolution, step/mm)
$102 = 80.000 (Z-axis travel resolution, step/mm)
$110 = 8000.000 (X-axis maximum rate, mm/min)
$111 = 8000.000 (Y-axis maximum rate, mm/min)
$112 = 3000.000 (Z-axis maximum rate, mm/min)
$120 = 250.000 (X-axis acceleration, mm/sec^2)
$121 = 250.000 (Y-axis acceleration, mm/sec^2)
$122 = 250.000 (Z-axis acceleration, mm/sec^2)
$130 = 264.000 (X-axis maximum travel, millimeters)
$131 = 264.000 (Y-axis maximum travel, millimeters)
$132 = 69.000 (Z-axis maximum travel, millimeters)

[image]

jahnj0584, 2017-06-07:
Do you mean when you hold down a jog button?

clarkdv, 2017-06-07:
Yes, when I click the X-, X+, Y-, Y+ buttons on the UGS Platform GUI or when I press the arrows on the keyboard. Slow as molasses.

winder, 2017-06-07:
@clarkdv you could try increasing your acceleration settings `$120`, `$121` and `$122`.

jahnj0584, 2017-06-07:
250mm is pretty fast for such a small machine. I think he means the delay
between sending each command, otherwise he could just jog 200mm and hit
stop when needed

chamnit, 2017-06-07:
This isn't a Grbl issue, but rather a UGS issue. Closing.

closed 2017-06-07 by chamnit

clarkdv, 2017-06-07:
I flashed GRBL v0.9 to the Arduino and the jogging works at the correct speeds. I flashed v1.1f back and it's dog slow again.

winder, 2017-06-07:
@chamnit UGS will switch between the new jog mode and G0 commands depending on the GRBL version. So it's possibly a setting issue.

@clarkdv are you using the same version of UGS for both tests?

clarkdv, 2017-06-07:
Yes, I actually have tested it with both UGS and UGS Platform and it's the same each time. v0.9 is OK, v1.1 is not.

clarkdv, 2017-06-07:
I just downloaded and ran UGS v1.09 and it jogs correctly with GRBL v1.1 So UGS v2.0 nightly doesn't and UGS Platform doesn't.

winder, 2017-06-07:
@clarkdv you mentioned changing the feed rate changes the speed slightly, did you try setting it to 8000 (your max speed setting)?

clarkdv, 2017-06-07:
Yes. And I meant changing the feed rates in the UGS GUI, not in the Arduino's GRBL settings.

wmkoller, 2021-04-11:
olá estou com esse mesmo problema. conseguisse resolver , eu troquei o protocolo e funcionou o jog , porem o laser passou,  a ficar fraco, não resolveu.
[image]