gnea/grbl #210: Speed up Grbl with a 20MHz UNO?

109JB, 2017-06-07:
For another project I just had to order a 20 MHz crystal oscillator, and wound up ordering 10 of them since the price difference between 1 and 10 was about 90 cents once shipping was considered.  So, I am thinking about swapping out the crysyal in one or more of the many Arduino UNOs I have and seeing about getting Grbl running a bit faster.  I was wondering if anyone had tried this, and if so, if they had any suggestions or things to avoid?  Here is what I plan:

http://masteringarduino.blogspot.com/2013/11/fasterArduino.html

Since this isn't really a Grbl issue I will close this immediately so that it doesn't sit in the open issues discussions.

closed 2017-06-07 by 109JB

chamnit, 2017-06-07:
I've never done it, but I can tell you now that the SAMD21 version of Grbl is running 2x-3x faster across the board.

109JB, 2017-06-07:
@chamnit - You tease!!!  I can't wait to see what you have going for the ARM version.  On that note,  I know that you are targeting the SAMD21, but is there any board in particular?  What board are you using for testing?  I was thinking about picking up a SAMD21 soon.

As for the crystal change in the UNO,  I have one machine that I would probably just leave on the UNO platform regardless of the release of the new ARM version.  I don't necessarily need more step rate, but faster throughput for 3D CAM files with lots of short moves is my goal.

electrokean, 2017-06-07:
@109JB I haven't done this with with grbl, but I did do it on my own custom PCB that included an ATmega328 with a 20MHz crystal (plus an ARM chip running Linux).  The ATmega was driving WS2812 LEDs from data sent via a Linux python app, and it all benefited from the speed and "nicer" timing.
The main thing is creating the boards.txt entry and a custom bootloader, which are covered in your linked article.  My bootloader and boards.txt are here https://github.com/moorescloud/holidayduino
If you run into any difficulties, I can probably help out.

109JB, 2017-06-08:
@electrokean - Thanks. It seems pretty straightforward.  I wonder why the Arduino folks decided on a 16MHz crystal when the ATmega 328 data sheet indicates that 20MHz is no problem?

electrokean, 2017-06-08:
https://electronics.stackexchange.com/questions/33166/why-do-arduinos-ship-w-16mhz-crystal-instead-of-20mhz

JBSchueler, 2017-06-08:
I have gbrl 1.1 running on a Arduino Uno @ 24MHz without any problems.
Keep in mind you have to flash a bootloader to support 20MHz (main reason is the correct baudrate to upload sketches.
When the bootloader is correct flashed, you can compile the gbrl for that board if you have added a board which has the correct fuses and clock speed.

109JB, 2017-06-08:
@JBSchueler - Thanks.  I am aware of the bootloader needing to be changed as it is in the link in my first post.  

What I am really curious about is if you have to change anything in the Grbl code?  I took a quick look around and it looks like the code is written to account for CPU clock differences and I didn't see anything at a quick glance that would need to be changed.

JBSchueler, 2017-06-08:
Correct, nothing has te be changed at all at the code of grbl.

chamnit, 2017-06-08:
@109JB : Just about any SAMD21 board should work, because most of the pins are able to do almost everything needed. I've got it working on an AdaFruit Feather, even with the reduced pin count, but enough to run a 3-axis machine. A SparkFun SAMD21 dev board has all of the pins exposed and can support 4-axes just fine. Even more if you use the SPI pins for other things. However, they burned a sh!*ty boot loader on it and it takes forever to flash it. You can fix it by burning the most recent one with a programmer. The Arduino Zero is ok too and has its own USB-serial converter with their debugger chip. 

Two things will help. A USB-serial converter definitely helps make Grbl run more efficiently. I've tested with an FTDI or a CH2104 and both run great. An external EEPROM i2c chip (~$2) is supported and will help not wear out the flash with emulated EEPROM (also supported).

If things work out like I plan, the new HAL should make it easy to support different boards and pin layouts, but more on that later.

109JB, 2017-06-16:
Well I got the 20MHz crystals, finished my other project and swapped out the 16MHz ceramic resonator for a 20MHz crystal in one of my UNOs. Right now some things work and some don't..it appears that the atmega328 is working fine, but there is some kind of communication problem between it and the 16U2 USB to serial chip. I can only get it to work at 57600 baud, and even then it isn't 100% reliable. I don't have a ton of test equipment, so I resort to writing interface programs and deducing what is going on from there.  On several other baud rates it appears that some characters have a hard time getting through, and in particular the \n eol character.  So anyway I could use some advice from those that have gone before.  Here is what I have tried.

1. Tried modifying bootloader make file and boards.txt on my own.
2. Tried mods from @electrokean that he posted above
3. Tried a few other bootloader/boards.txt mods from others.

All of the above gave essentially the same results.

As for the hardware, I soldered in the crystal, but didn't have any 20pF caps so left them out. I read somewhere of success without them, but I have ordered some to put in and try. I live in the country so have no close place to get them.  Do you think this could be what is happening?

Also, I have used fuse setting of others and also used a fuse calc, starting with the default uno fuse setting and then only changing to get a full swing crystal oscillator.  Is there any other setting I can try.

As I said, the atmega328 seems to be working fine. I wrote some sketches and uploaded them using an external programmer and they do what they are supposed to.

Any help is appreciated

electrokean, 2017-06-18:
Interesting.  Yes I think this must be the missing load capacitors, as without them your crystal could be running way off frequency.  I had a sketch that used 115200 baud using 20MHz crystal, and I never saw any communications issues (but thinking back it mostly used SPI for data transfers).

X3msnake, 2017-06-18:
Dont you have caps that you can serial mount to add up to the required
capacitance?

electrokean, 2017-06-19:
Or "borrow" some from another device (grin)

109JB, 2017-06-19:
This is kind of a side project anyway, so although I could liberate some caps from some other junk device, I will just wait for the ones I ordered to come in. I have too many other projects going on anyway.

X3msnake, 2017-07-02:
Cool keep us posted on the results succeses or failures ;)

mkeyno, 2017-07-09:
has anyone tested heavy duty job like laser engraving with this new serial speed?

109JB, 2017-07-10:
@mkeyno 

This has absolutely nothing to do with serial speed. Serial speed is controlled by a separate USB to serial converter chip and it has its own crystal. This modification is for the crystal controlling the 328P microprocessor.

As a side note, I got my caps a while ago but other things have presented me getting them put in.

X3msnake, 2017-07-10:
Cool keep us posted when you get around to do it ;)