gnea/grbl #211: Path Stepper Error

jascotty2, 2017-06-11:
I've found an issue with one specific set of coordinates, and spent a great deal of time with the hardware and trawling the code trying to nail it down: When moving relative X1.7 Y8 (+/- .1), the x axis doesn't move correctly - it vibrates, and moves about half of the required distance, if at all. Similarly, X8 Y1.7 does the same thing to the y axis.

I eventually discovered that setting the x and y max rates to different speeds was the culprit.

I haven't figured out where in the code might be responsible for the problem, or even if there is some obscure electrical issue with my setup, but I wanted to put this out there, in case this is a reproducible bug.

Settings:
$100=4024
$101=4024
$102=4024
$110=11
$111=13
$112=13

jahnj0584, 2017-06-11:
Sounds more like you have mechanical binding issues in the machine there.

4000 steps/mm is also crazy high.... Why is it so?

jascotty2, 2017-06-11:
I calibrated it for inches, instead of mm ;)

I'm not sure if it's binding - faster and slower step speeds work fine, and faster speeds have less torque..

electrokean, 2017-06-11:
This sounds awfully like mid-band resonance in your stepper motors, or just pushing your motors too hard.  It is very unlikely to be an issue with the grbl code.  You may be able to get around it by adjusting parameters such as max step or acceleration, but it is more an electro-mechanical issue.

Some info that might be useful in assisting:
- motor size & specs
- stepper driver specs
- driver supply voltage
- motor current setting
- micro stepping setting
- machine style & drive system (mill/gantry, belt/acme/ball screws, etc).

First thing I'd suggest trying if possible is adjusting (increasing) micro-stepping, and increasing your driver voltage.  But this is really beyond the scope of issue tracking for grbl, and you should probably be posting on a site like cnczone.

chamnit, 2017-06-11:
Your settings seem weird with a max rate of 11 and 13 mm/min. All Grbl settings are in terms of mm, not and never inches. It would help if you supplied all of your $$ Grbl settings and the set of gcode commands you are using that produces your problem. 

Also make sure you are below the 30 kHz max step rate Grbl supports.

chamnit, 2017-06-15:
Closing due to no response.

closed 2017-06-15 by chamnit