gnea/grbl #214: Homing and positive working space....again

mtthamann, 2017-06-14:
Hello there,

I am afraid I am asking the same stupid question every newbie does, but I combed different boards to find a solution, I read the FAQ and tried for three days, so please be patient with me ;)

I build my very first CNC machine, a gantry style laser engraver, modelled after an example at instructables. Astonishingly it works quite fine, the mechanics are ok, the general functions are working. I can do basic patterns and lines in manual mode (entering G-code by keyboard), the PWM for the laser works great and the first results are making me happy. The layout is according to right hand rule, so X positives moves the gantry to the right, Y should move away from me.

What is driving me crazy is the use of homing function and positive coordinates to work with LaserGRBL.
I managed to invert my homing sequence to the negative side of the working space, so homing is now bottom left of my table. The strange thing is
- the X-axis works now in positive numbers, so G90 G0 X10 moves the gantry 10mm over to the right
- the Y-axis works in negative numbers, so G90 G0 Y-10 is needed to move the laser away from me.

If I try G90 G0 Y10 the system tells me directly I am outside the soft limits and quits with an error and alarm. This makes me think I have set something set up wrong as the systems knows it must move to negative direction.
This is my settings:
[image]
I uncommented HOMING_FORCE_SET_ORIGIN in config.h before compiling , as well  LIMITS_TWO_SWITCHES_ON_AXES

Sorry if anything is unclear, english is not my first language (greetings from Germany by the way).
Thank you very much for the work on this software, it is a real fun project to my son (13yo) and me to build this machine.
My background is not electrical but an engineer (processing, furnaces mostly) so I have an technical background, even though using a mill or other machines is 20 years gone. I tried relearning about coordinate systems and right hand rule.

Any help appreciated!

neilferreri, 2017-06-15:
@mtthamann  Send a "$3=2" to invert the Y-axis. The standard homing is upper right, but it really doesn't matter where you home as long as you do it.

chamnit, 2017-06-15:
@mtthamann : You want to use G91, not G90, to test if the axes move in the correct directions. G91 is incremental mode and will move relative to the last programmed point.

mtthamann, 2017-06-16:
@neilferreri 
@chamnit 
Thanks for the help, I appreciate it very much and I tried both.
Sending a $3=2 to the system changes the Y homing position to the other side of the machine, as I expected. Nevertheless is the working space then still in negative direction. If I send any command moving the Y axis to positive direction I get no movement and an instant feedback: soft limit.
Wether I test it with G91 or G90 movements, it doesn´t change the result.
The X axis workspace is all positive numbers
The Y axis workspace is all negative numbers.

Any other ideas? Thank you very much!

mtthamann, 2017-06-16:
Just to show my problem I get with LaserGRBL. If I start any program created from a picture the system does the same start sequence.
[image]
I my understanding it sets the system to G90 (coordinates from starting point
F500 The system sets the force I put in
G0 X10 Y10 The start point, depending from my offset I put in, is set

This already triggers the Soft limit reaction three lines down, because LaserGRBL is trying to reach a point which is not within the working space, as Y-axis is all negative.
Hope this clarifys my problem.

neilferreri, 2017-06-16:
@mtthamann Maybe this is it: You have inverted your X axis homing, but not the Y. Try inferring your Y direction as I stated before and also inverting it for your homing sequence. So $3=2 inverts the Y direction and $23=3 inverts homing direction for both X and Y. Your machine "thinks" it is at the front right corner.

109JB, 2017-06-16:
Where the machine homes makes absolutely no difference. I can set my machine up to home right in the middle of the travels and can still get it to work.  Homing is just a reference position.

As for all coordinates being negative.  That is only in machine space anyway so who really cares. Unless you send a G53 on every line you aren't using machine coordinates anyway.  There are 6 coordinate system workspaces (G54 - G59) that can/are offset from the machine coordinate system and you are always using one of them.  G54 is the default and when you first load Grbl the offsets for all of them are at 0. Since the default offsets are 0 that means that to begin with all of those coordinate workspaces are essentially the same, but you can set each one up however you want.

Just position the axes where you want them to be zero and send a G10 L20 command to set that position to zero. For example to zero the X axis you would send G10 L20 P1 X0. The G10 L20 part tells that you are going to set a coordinate system. P1 is the coordinate system where P1 - P6 relates to G54 - G59.  Finally the X0 is the value you want the X axis set to. You can include all axes on one line if you want. So, a G10 L20 P3 X1 Y2 Z3  would set the current position of the machine as X1, Y2, Z3 in the G56 coordinate system

mtthamann, 2017-06-16:
@neilferreri 
That was it, you were right.
I didn´t get that $23 can be set for each axis individually, I thought it was only an 0 / 1 decission wether to invert or not.
So I did set $23=3 and $3=2 and it works perfect.
THANK YOU VERY MUCH!

@109JB 
Thanks for the explanation, I will keep this in mind for future use, i.e. when using different starting points on smaller workingpieces.

closed 2017-06-16 by mtthamann