gnea/grbl #215: Updating GRBL on Woodpecker CNC Board

Deckard-BR, 2017-06-20:
Hi,
I have just got a 2418 CNC machine up an running. 

It has a Woodpecker Control Board with GRBL 0.9j installed. I would like to update it to GRBL 1.1f.

So far everything I have tried (Arduino IDE, loading HEX file etc) has failed.

I was wondering if anyone has had success in uploading to this board ?

And, if so, how...

T.I.A.

109JB, 2017-06-20:
About 30 seconds search on Google.

http://delta.superhosting.bg/~ermacon5/longgo/?x=entry:entry170422-235212

Deckard-BR, 2017-06-20:
Thanks for pointing me in that direction. And thanks to longgo for a very clear explanation. It was the way I was heading, but hearing from someone else who has done it makes all the difference.

closed 2017-06-20 by chamnit

ablankenship10, 2021-01-06:
For future reference it is good to include the full answer in text as website links may go offline, for those that come by 4 years later :)

Ronald666666, 2025-07-09:
> For future reference it is good to include the full answer in text as website links may go offline, for those that come by 4 years later :)

I come from the future ... this indeed happened :( ... Does anyone has a working link?