gnea/grbl #216: 1.1e eight short lines burning 45 degree angle problem

rfresh737, 2017-06-21:
I have a g-code file that burns 8 one inch long lines at a 45 degree angle, each one next to each other. The laser burns these lines such that they look like each 2 line set are next to each other, so it looks like there are 4 sets of two lines instead of 8 lines evenly spaced.

I'm attaching the file. This file looks OK in two different g-code simulators I have input the file into. But when actually laser burning this file onto craft plywood, the lines are not evenly spaced. Please see attachments.
Same results using Grbl 0.9i. I normally use Grbl 1.1e.

[image]

[45LinesBiDirectional.zip](https://github.com/gnea/grbl/files/1090361/45LinesBiDirectional.zip)

dungjk, 2017-06-21:
Please post your machine configurations and distance between lines.
This iuusue may cause by following reasons:
* Hardware issue:
** Un-tight timing belt - check to tensioning it
** Friction on slider keep x/y axis to move until the traction high enough to move - use oil and tensioning the belt.
** Timing belt to soft, it's elastic make the machine move incorrectly. Suggest to use similar type with steel core (https://aliexpress.com/store/product/PU-with-steel-core-GT2-belt-2GT-timing-belt-6mm-width-10m-a-pack-for-3d/1962255_32547444991.html?isOrigTitle=true)
* Software issue:
** Machine's configuration steps per mm and distance between lines, when machine rounding move distance to steps it are not fit in same (You can see that other lines with different distance are burning in correct distance). I used to have same issue with 20 teeth gt2 pulley then i change to use 16 teeths gt2 pulley and change machine configs, after that the issue disappear.
** Software is used to generate gcode, it rounding the output values (this is rarely happened but one of my friend got it. He try to change the software and it work ok).

mayhem2408, 2017-06-21:
I have seen this before. As the Laser head is changing direction, in your case on the X axis, it appears there might be some back lash. This is more evident on belt driven systems and as @dungjk has mentioned could be a soft rubber belt and the belt is not tensioned enough. A simple test that might show the problem is to enable the steppers so the carriage is locked in place and try to move the laser carriage. It should not more at all. If it can move, you have backlash.

rfresh737, 2017-06-21:
The belts had some give to them, so they were not what I would call 'tight', but I loosen them up a little bit and now the 45 degree lines burn evenly.

Thank you all.

closed 2017-06-21 by chamnit