gnea/grbl #218: Pausing g-code job?

rfresh737, 2017-06-22:
I'm using Grbl 1.1e. Is there a Grbl command that I can insert into my g-code to pause the job for a certain number of seconds and then have it resume?

I was looking at online references such as: G04 P500

...but did not see this (or missed it) for Grbl 1.1e.

Thanks...

chamnit, 2017-06-22:
Says its supported right [here](https://github.com/gnea/grbl/wiki) and the bottom of the page [here](https://github.com/gnea/grbl). This [page](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Commands) has only shows the modal groups supported by Grbl, which does not include the non-modal commands like G4, G53, etc. You probably saw this page. I'll add something there for completeness.

closed 2017-06-22 by chamnit

rfresh737, 2017-06-22:
What is the format for the argument? is P500 a 1/2 second delay?

chamnit, 2017-06-22:
On this [page](https://github.com/gnea/grbl/wiki), there is a [link](http://linuxcnc.org/docs/html/gcode.html) to LinuxCNC's g-code descriptions. Grbl follows everything there, except for a handful of special commands, as stated on that same page.