gnea/grbl #220: I can't move With limit switches enabled, but with disabled, there's no problem

norbigazda98, 2017-07-04:
Hi,
I have  a problem. 
_I installed 6 limit switches to the ends of each axes. I had interference problems. I was told to do [this](https://github.com/gnea/grbl/wiki/Wiring-Limit-Switches). With improved noise filtering, so with the resistors and capacitors. 
I did it, and the movements now were longer than a few millimeter, before it froze. 
I was told to use homing. That didn't work either. I was told, that it's because the max travel is wrong. I fixed that as well._ 

So here is the problem: 
When I enable the homing and the limit switches, set up that they are normally open, and start a homing cycle, it finishes without any problems. But after that when I move the axes, after a few movements it comes to a complete halt and tels me to soft reset. 
When no homing and no limit switches are enabled, everything works just fine. 

How can I make it work?

vMeph, 2017-07-04:
sounds like you still have noise issues, i use to have same problems i fixed my problems by using NC switches and put shield cables in all machine.
do you have soft limits also enable?
if so try disble them and give it a test

chamnit, 2017-07-04:
Please post your settings, Grbl version, and the error string. It will help us diagonose your problem much faster.

electrokean, 2017-07-05:
@norbigazda98 Yes, still sounds like a noise issue.
When you move after homing and it comes to a halt, what message/alarm do you get? (other than soft reset)
Does it happen without the spindle running?  I'm assuming this is a CNC router.

norbigazda98, 2017-07-05:
After finishing the honing cycle and trying to move and it stops, the only things that happen is that the active status is Alarm!, so alarm is triggered and it tells me: #date# [MSG: Reset to continue] 

Nothing else happens. If I reset, it works again until it freezes like this the next time. 

I'll post the settings soon, but not from the mobile site. 
Soft limits are not enabled.

X3msnake, 2017-07-05:
Most likely noise

Check the wiki
https://github.com/gnea/grbl/wiki/Wiring-Limit-Switches

norbigazda98, 2017-07-05:
That's what I did. I made the wirings that has been described there and it still does this. 

And also, if the psu for the drivers is turned off, (so the motors don't move meanwhile grbl thinks they do) there's no freeze. The psu is an inverter. Is that the problem? Should I try to move it even further? It's half a meter from the arduino board.

norbigazda98, 2017-07-05:
[image]
Here are the settings

X3msnake, 2017-07-05:
I assume you are using the capacitor decoupling method right?

Can you try the optoisolation method?

electrokean, 2017-07-05:
A photo of your machine and wiring is probably in order.  We may be able to make suggestions to improve noise immunity.

vMeph, 2017-07-05:
also sounds like your settings for max rate is beyond the max capacity Khz that arduino can handle that is 30Khz 
right now you runing at 41Khz

norbigazda98, 2017-07-05:
Yes, it's the capacitor method. I don't think that I can try the opto method soon, maybe in weeks :( 

I'll make a photo 

vMeph: What do you mean?

vMeph, 2017-07-05:
@norbigazda98 never mind wrong math

X3msnake, 2017-07-05:
Can you try passing the PSU Cables that power the drivers trough a ferrite?

chamnit, 2017-07-10:
Given this is likely an electrical issue, not Grbl firmware, this issue is closed. But we'll continue to help you out. I'm trying to keep the issues threads at this site under control.

closed 2017-07-10 by chamnit

norbigazda98, 2017-07-10:
I'm on a holiday, that's why I didn't write. 

So we can continue the conversation here, but now the topic is titled closed, if I understand correctly?

X3msnake, 2017-07-10:
yes

norbigazda98, 2017-07-17:
Here is the machine, but of course it's not final. 
[image]
[image]
[image]