gnea/grbl #221: How to prevent stepper overheating when in constant torque

Mistranger, 2017-07-05:
I have a gcode program for my laser engraver that uses X movement a lot but very little Y movement (raster engraving), which makes Y stepper become very hot when in holding torque mode. How to tell grbl to disable Y stepper when it's not moving? (so it won't consume power and heat up)

electrokean, 2017-07-05:
You generally don't want to disable the stepper driver as you can lose position, especially if using micro-stepping.  And grbl has no way to know or control this on a per axis basis anyway.
Power consumption of the steppers when stationary (and thus the heat produced by them) is a function of the square of the current, so a small reduction can make a noticeable difference.  Some of the better stepper drivers do reduce the motor current when idle.  Or if you know you're going to run one of these jobs, you could manually reduce the current setting on your stepper driver.  Just ensure it is still sufficient to not lose steps during the few moves.

X3msnake, 2017-07-05:
You should check that the current set in your driver chip is adequate for
your motor. Check motor datasheet to see the correct amps and the driver
chip's datasheet to see how to set the correct current.

What drivers are you using drv8825?
What motors do you have?

109JB, 2017-07-05:
Most steppers are rated for continuous use with case temperatures of 100 deg.C.  Many people think their motors are getting too hot because they are too hot to touch, but the motors are made to take it. The motor can be well below its maximum temperature and still "feel" hot.

chamnit, 2017-07-05:
@Mistranger : You have all good and relevant answers. Closing.

closed 2017-07-05 by chamnit