gnea/grbl #223: Laser mode feed limit?

Mistranger, 2017-07-06:
I'm engraving 2 different images using GRBL 1.1 laser mode. First image's resolution is 5 dots/mm, so its gcode looks like this:
```
G1 X27  S244
G1 X26.8  S313
G1 X26.6  S357
G1 X26.4  S327
G1 X26.2  S202
G1 X26  S76
G1 X25.8  S72
G1 X25.6  S0
G1 X25.4  S134
G1 X25.2  S218
G1 X25  S114
G1 X24.8  S0
G1 Y130.6 
G1 X23.8 
G1 X24  S76
G1 X24.2  S0
G1 X25.2 
G1 X25.4  S194
G1 X25.6  S222
G1 X25.8  S168
G1 X26  S204
G1 X26.2  S158
G1 X26.4  S150
G1 X26.6  S128
G1 X26.8  S82
G1 X27  S172
G1 X27.2  S265
```
Second image's resolution is 10 dots/mm, and here is the example of its gcode:
```
X178S677
X178.1S693
X178.2S699
X178.3S693
X178.4
X178.5S699
X178.6S705
X178.7S718
X178.8S734
X178.9S746
X179S756
X179.1S762
X179.2S765
X179.3
X179.4S759
X179.5S752
X179.6S743
X179.7
X179.8
X179.9S740
Y124.4S737
X179.8S740
X179.7S749
X179.6S752
X179.5S759
X179.4S762
X179.3S768
X179.2S771
X179.1S768
X179
X178.9S759
X178.8S746
X178.7S730
X178.6S715
X178.5S699
X178.4S696
X178.3
X178.2S693
X178.1
X178S705
X177.9S696
X177.8S683
X177.7S677
X177.6S683
```

I set X axis speed to 2000mm/min in GRBL settings and both gcode programs use feed rate 2000mm/min (F2000), but GRBL runs them much slower and in case of first program the actual feed rate is twice faster than when running the second job. Is this the limit of GRBL laser mode, or I'm doing something wrong?

109JB, 2017-07-06:
Please post your Grbl settings from a $$ request.

chamnit, 2017-07-06:
@Mistranger : This is an performance limit. When running at higher feed rates and decreasing the dots/mm, you end up sending flooding Grbl with more and more communication data to execute the job. At a certain point, Grbl can't receive, parse, process, and execute a single g-code line fast enough to keep the planner full and executing efficiently. This limit is around 4ms, sometimes more depending on the job. 

Believe it or not, this is a common issue with all existing firmwares. There are a lot of bottlenecks, some CPU speed, others communication overhead, code efficiency, etc. Grbl actually performs very well compared to others with on much faster chips due to code efficiency and its communication protocol. Grbl-LPC, a Grbl port onto Smoothieboard, blazes on laser raster jobs. AFAIK, it outperforms most, if not all, open-source firmwares in speed. I do have plans to solve some of the other bottleneck problems, but they will only be in the next version of Grbl, which will be on an ARM chip and the Mega2560 only.

closed 2017-07-06 by chamnit