gnea/grbl #224: GRBL does not respond to commands as expected after an ALARM

winder, 2017-07-08:
When GRBL enters an unexpected ALARM state such as when exceeding the soft limits, all messages besides a soft reset are ignored.

My expected behavior is that GRBL should behave the same way as when starting with homing enabled. The state is set to `Alarm`, status events indicate the state as `Alarm`, and commands return an `error:`.

chamnit, 2017-07-08:
This is the intended behavior (mostly). See the interface (doc)[https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface#feedback-messages]. Originally, I meant to allow status reports during a critical alarm, but forgot to add it before releasing v1.1. At this point, I can't alter this until the next version.

closed 2017-07-08 by chamnit

ashwath72, 2017-07-14:
Hi am ASHWATH from Bangalore, India  Please help me, Bro, I built mini CNC plotter machine at home but this is not Working pls help me. I am using Arduino, L293d Motor shield old 2nos DVD drive Micro Servo Stepper Motor it's not working one time its works I am very happy pls pls pls pls help me, Bro Please send me any attaching links ASAP. waiting for your reply

houguang, 2017-07-14:
@ashwath72 Specifically talk about how not to work