gnea/grbl #225: Setting WCO to zero

andrescamarena, 2017-07-09:
Hello,

I'm trying to get by WCO coords to be zero when I set WPos coords to be zero. But I don't seem to be able to figure it out. Tried using 38.1 to set to tool offset to zero and with G10/G92. But I can't get WCO position to change to (0,0,0). What is the correct way to tell the machine the current location is WCO = (0,0,0) and WPos =(0,0,0)?

Thank you!
Andres

vMeph, 2017-07-09:
you want to perform a G54 to make work coords to be 0
G10 P1 L20 X0 Y0 Z0  

the P is where you set
P1 = G54
P2 = G55
P3 = G56
P4 = G57
P5 = G58
P6 = G59

andrescamarena, 2017-07-09:
I tried this, but wasn't able to get WCO (0,0,0) 

G54
 G54Z0
 G10P1L20X0Y0Z0
 G54Z0
 G54
<Idle|WPos:0.000,0.000,0.000|FS:0,0|WCO:-91.000,-96.001,-9.667>

vMeph, 2017-07-09:
you go position you want the WCO to be and set
G10 P1 L20 X0 Y0 Z0
then when you do a job you put on file 
G54
so when  when you are at home

chamnit, 2017-07-09:
To set the offsets to zero (work coordinates will be equal to machine coordinates), use G10 L2, not G10 L20. L2 sets the offset values directly. L20 sets the offset values to what you want the current position to be.

closed 2017-07-09 by chamnit

langwadt, 2017-07-09:
To get  WCO = (0,0,0) and WPos =(0,0,0), MPos needs to be set to (0,0,0) afaik only reset can do that

vMeph, 2017-07-10:
@chamnit  sorry to bother you with this questions but  you say a G54 is set by use G10 L2 
so lets say i sens machine home, with my pull of switch i get Mpos X-10 Y-10 Z-10 Wco X-10 Y-10 Z-10
I move to desire start point to set Wco     Mpos X-100 Y-100 Z-20 Wco X-100 Y-100 Z-100
I press G10 L2 P1 X0 Y0 Z0  and i get same Mpos X-100 Y-100 Z-20 Wco X-100 Y-100 Z-100
If i do same but using G10 L29 P1 X0 Y0 Z0 then i get  Mpos X-100 Y-100 Z-20 Wco X0 Y0 Z0

The G10 L2 P1 X0 Y0 Z0  nothing changes is there something im missing?

chamnit, 2017-07-10:
@vMeph : It is explained in detail [here](http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g10-l2) and [here](http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g10-l20). 

In your L2 example, you should get WCO: 0,0,0 and WPos:  X-100 Y-100 Z-20.
In your L20 example, you should get WCO: 100,100,20 (might be negative) and WPos:0,0,0.

And @langwadt  is correct. To have WCO = 0,0,0 and WPos= 0,0,0, MPos has to be 0,0,0 as well. Grbl's homing cycle almost never finishes this way unless you have HOMING_FORCE_ORIGIN enabled or pull-off motion to zero(may not allow you do this).

vMeph, 2017-07-11:
@chamnit you correct WCO is set by G10 L2 P1
In case G10 L2 P1X0 Y0 Z0 will clean the WCO
I guess the G10 L20 P1 will set the Wpos

Was wondering about the $G modal i only see there the G54, how to add more $G modal lines?
Is this done by using the $N1?

Thanks

chamnit, 2017-07-11:
@vMeph : I'm not sure what you mean by adding more $G modal lines. Gcode modes are explicitly defined by the standard. If you mean if you want more work coordinate systems, you have six available, G54 - G59. Set by G10 P1 to P6, respectively. The `$#` command will show you the offsets for all of them.