gnea/grbl #226: 断电如何确定xy轴位置

houguang, 2017-07-10:
我生成的两个nc文件,一个是粗加工,一个精加工,粗加工完断电换刀,再通电后,xy轴位置与粗加工时的位置对不上,请问这个问题有什么好的解决办法.

cheton, 2017-07-10:
**Please use English to post issues here so others can respond to your question.**

Title: How to find the zero reference point of my workpiece when returning from a power-off state.

Each time you turned off and then turned on the power, you can use a so-called Edge Finder to zero your workpiece, one of the cheapest solution is the aluminium foil tape (铝箔胶带), you can check it out here: http://www.instructables.com/id/Poor-Mans-CNC-Edge-Finder/

chamnit, 2017-07-10:
Or use homing switches and work coordinate systems to set part zero. This is exactly what they are for.

closed 2017-07-10 by chamnit

houguang, 2017-07-11:
@cheton Thank you for your answer, but arduino cnc did not find the edge function. How does the operation. (This is Google translation, I do not speak English)