gnea/grbl #228: Error messages when compiling grbl 1.1e with Arduino IDE 1.8.3

tingox, 2017-07-11:
When I compile / upload grbl 1.1e from Arduino IDE 1.8.3, I get some error messages. The upload finishes and grbl appears to work afterwards (I can send commands, and grbl responds, the response looks ok).
Note: I had to copy the 'grbl' folder from the unzipped 1.1e release zip file manually, because the automated versions "Manage Libraries" and "Add .ZIP Library" only produced an error message.
Error messages from compiling grbl:
```
In file included from /home/tingo/Arduino/libraries/grbl/grbl.h:43:0,
                 from /home/tingo/work/bitraf/2017/20170710_grbl/grbl/grbl-1.1e.20170114/grbl/examples/grblUpload/grblUpload.ino:27:
/home/tingo/Arduino/libraries/grbl/nuts_bolts.h:25:0: warning: "false" redefined
 #define false 0
 ^
In file included from /zs/tingo/progs_lin/arduino-1.8.3/hardware/arduino/avr/cores/arduino/Arduino.h:24:0,
                 from sketch/grblUpload.ino.cpp:1:
/zs/tingo/progs_lin/arduino-1.8.3/hardware/tools/avr/lib/gcc/avr/4.9.2/include/stdbool.h:42:0: note: this is the location of the previous definition
 #define false false
 ^
In file included from /home/tingo/Arduino/libraries/grbl/grbl.h:43:0,
                 from /home/tingo/work/bitraf/2017/20170710_grbl/grbl/grbl-1.1e.20170114/grbl/examples/grblUpload/grblUpload.ino:27:
/home/tingo/Arduino/libraries/grbl/nuts_bolts.h:26:0: warning: "true" redefined
 #define true 1
 ^
In file included from /zs/tingo/progs_lin/arduino-1.8.3/hardware/arduino/avr/cores/arduino/Arduino.h:24:0,
                 from sketch/grblUpload.ino.cpp:1:
/zs/tingo/progs_lin/arduino-1.8.3/hardware/tools/avr/lib/gcc/avr/4.9.2/include/stdbool.h:43:0: note: this is the location of the previous definition
 #define true true
 ^
In file included from /home/tingo/Arduino/libraries/grbl/grbl.h:43:0,
                 from /home/tingo/work/bitraf/2017/20170710_grbl/grbl/grbl-1.1e.20170114/grbl/examples/grblUpload/grblUpload.ino:27:
/home/tingo/Arduino/libraries/grbl/nuts_bolts.h:56:0: warning: "max" redefined
 #define max(a,b) (((a) > (b)) ? (a) : (b))
 ^
In file included from sketch/grblUpload.ino.cpp:1:0:
/zs/tingo/progs_lin/arduino-1.8.3/hardware/arduino/avr/cores/arduino/Arduino.h:93:0: note: this is the location of the previous definition
 #define max(a,b) ((a)>(b)?(a):(b))
 ^
In file included from /home/tingo/Arduino/libraries/grbl/grbl.h:43:0,
                 from /home/tingo/work/bitraf/2017/20170710_grbl/grbl/grbl-1.1e.20170114/grbl/examples/grblUpload/grblUpload.ino:27:
/home/tingo/Arduino/libraries/grbl/nuts_bolts.h:57:0: warning: "min" redefined
 #define min(a,b) (((a) < (b)) ? (a) : (b))
 ^
In file included from sketch/grblUpload.ino.cpp:1:0:
/zs/tingo/progs_lin/arduino-1.8.3/hardware/arduino/avr/cores/arduino/Arduino.h:92:0: note: this is the location of the previous definition
 #define min(a,b) ((a)<(b)?(a):(b))
 ^
In file included from /home/tingo/Arduino/libraries/grbl/grbl.h:43:0,
                 from /home/tingo/work/bitraf/2017/20170710_grbl/grbl/grbl-1.1e.20170114/grbl/examples/grblUpload/grblUpload.ino:27:
/home/tingo/Arduino/libraries/grbl/nuts_bolts.h:61:0: warning: "bit" redefined
 #define bit(n) (1 << n)
 ^
In file included from sketch/grblUpload.ino.cpp:1:0:
/zs/tingo/progs_lin/arduino-1.8.3/hardware/arduino/avr/cores/arduino/Arduino.h:123:0: note: this is the location of the previous definition
 #define bit(b) (1UL << (b))
 ^
Sketch uses 30320 bytes (93%) of program storage space. Maximum is 32256 bytes.
Global variables use 1631 bytes (79%) of dynamic memory, leaving 417 bytes for local variables. Maximum is 2048 bytes.
Low memory available, stability problems may occur.
```
my machine runs Fedora 25:
    [tingo@kg-elitebook ~]$ uname -a
    Linux kg-elitebook.kg4.no 4.11.8-200.fc25.x86_64 #1 SMP Thu Jun 29 16:13:56 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux
Hope this helps

chamnit, 2017-07-11:
Looks like you have multiple Grbl installations. One in a working directory and the other in the Arduino library.

tingox, 2017-07-11:
I don't think that is the issue. I tried again, this time loading grblUpload from the libraries directory:
```
In file included from /home/tingo/Arduino/libraries/grbl/grbl.h:43:0,
                 from /home/tingo/Arduino/libraries/grbl/examples/grblUpload/grblUpload.ino:27:
/home/tingo/Arduino/libraries/grbl/nuts_bolts.h:25:0: warning: "false" redefined
 #define false 0
 ^
In file included from /zs/tingo/progs_lin/arduino-1.8.3/hardware/arduino/avr/cores/arduino/Arduino.h:24:0,
                 from sketch/grblUpload.ino.cpp:1:
/zs/tingo/progs_lin/arduino-1.8.3/hardware/tools/avr/lib/gcc/avr/4.9.2/include/stdbool.h:42:0: note: this is the location of the previous definition
 #define false false
 ^
In file included from /home/tingo/Arduino/libraries/grbl/grbl.h:43:0,
                 from /home/tingo/Arduino/libraries/grbl/examples/grblUpload/grblUpload.ino:27:
/home/tingo/Arduino/libraries/grbl/nuts_bolts.h:26:0: warning: "true" redefined
 #define true 1
 ^
In file included from /zs/tingo/progs_lin/arduino-1.8.3/hardware/arduino/avr/cores/arduino/Arduino.h:24:0,
                 from sketch/grblUpload.ino.cpp:1:
/zs/tingo/progs_lin/arduino-1.8.3/hardware/tools/avr/lib/gcc/avr/4.9.2/include/stdbool.h:43:0: note: this is the location of the previous definition
 #define true true
 ^
In file included from /home/tingo/Arduino/libraries/grbl/grbl.h:43:0,
                 from /home/tingo/Arduino/libraries/grbl/examples/grblUpload/grblUpload.ino:27:
/home/tingo/Arduino/libraries/grbl/nuts_bolts.h:56:0: warning: "max" redefined
 #define max(a,b) (((a) > (b)) ? (a) : (b))
 ^
In file included from sketch/grblUpload.ino.cpp:1:0:
/zs/tingo/progs_lin/arduino-1.8.3/hardware/arduino/avr/cores/arduino/Arduino.h:93:0: note: this is the location of the previous definition
 #define max(a,b) ((a)>(b)?(a):(b))
 ^
In file included from /home/tingo/Arduino/libraries/grbl/grbl.h:43:0,
                 from /home/tingo/Arduino/libraries/grbl/examples/grblUpload/grblUpload.ino:27:
/home/tingo/Arduino/libraries/grbl/nuts_bolts.h:57:0: warning: "min" redefined
 #define min(a,b) (((a) < (b)) ? (a) : (b))
 ^
In file included from sketch/grblUpload.ino.cpp:1:0:
/zs/tingo/progs_lin/arduino-1.8.3/hardware/arduino/avr/cores/arduino/Arduino.h:92:0: note: this is the location of the previous definition
 #define min(a,b) ((a)<(b)?(a):(b))
 ^
In file included from /home/tingo/Arduino/libraries/grbl/grbl.h:43:0,
                 from /home/tingo/Arduino/libraries/grbl/examples/grblUpload/grblUpload.ino:27:
/home/tingo/Arduino/libraries/grbl/nuts_bolts.h:61:0: warning: "bit" redefined
 #define bit(n) (1 << n)
 ^
In file included from sketch/grblUpload.ino.cpp:1:0:
/zs/tingo/progs_lin/arduino-1.8.3/hardware/arduino/avr/cores/arduino/Arduino.h:123:0: note: this is the location of the previous definition
 #define bit(b) (1UL << (b))
 ^
Sketch uses 30320 bytes (93%) of program storage space. Maximum is 32256 bytes.
Global variables use 1631 bytes (79%) of dynamic memory, leaving 417 bytes for local variables. Maximum is 2048 bytes.
Low memory available, stability problems may occur.
```
still those error messages.

chamnit, 2017-07-11:
This type of error can also occur if you have altered the Grbl example upload sketch in anyway. It should only have a include line for the grbl.h file. Again, please make sure all Grbl source files are removed from all Arduino search paths. There can only be one.

tingox, 2017-07-11:
the grblUpload sketch is unaltered.
I removed everything (which was only the grbl folder) from Arduino IDE libraries:
```
[tingo@kg-elitebook ~]$ ls -la ~/Arduino/libraries/
total 12
drwxrwxr-x. 2 tingo tingo 4096 Jul 11 18:13 .
drwxrwxr-x. 3 tingo tingo 4096 Jul 11 15:50 ..
-rw-rw-r--. 1 tingo tingo   87 Jul 11 15:50 readme.txt
```
then I removed the contents of the Arduino IDE preferences directory
```
[tingo@kg-elitebook ~]$ ls -la ~/.arduino15/
total 8
drwxrwxr-x.  2 tingo tingo 4096 Jul 11 18:18 .
drwx------. 30 tingo tingo 4096 Jul 11 17:09 ..
```
Next I copied the grbl folder back into the Arduino IDE libraries folder again, and started Arduino IDE. Opened the grblUpload sketch, and pressed 'upload':
```
In file included from /home/tingo/Arduino/libraries/grbl/grbl.h:43:0,
                 from /home/tingo/Arduino/libraries/grbl/examples/grblUpload/grblUpload.ino:27:
/home/tingo/Arduino/libraries/grbl/nuts_bolts.h:25:0: warning: "false" redefined
 #define false 0
 ^
In file included from /zs/tingo/progs_lin/arduino-1.8.3/hardware/arduino/avr/cores/arduino/Arduino.h:24:0,
                 from sketch/grblUpload.ino.cpp:1:
/zs/tingo/progs_lin/arduino-1.8.3/hardware/tools/avr/lib/gcc/avr/4.9.2/include/stdbool.h:42:0: note: this is the location of the previous definition
 #define false false
 ^
In file included from /home/tingo/Arduino/libraries/grbl/grbl.h:43:0,
                 from /home/tingo/Arduino/libraries/grbl/examples/grblUpload/grblUpload.ino:27:
/home/tingo/Arduino/libraries/grbl/nuts_bolts.h:26:0: warning: "true" redefined
 #define true 1
 ^
In file included from /zs/tingo/progs_lin/arduino-1.8.3/hardware/arduino/avr/cores/arduino/Arduino.h:24:0,
                 from sketch/grblUpload.ino.cpp:1:
/zs/tingo/progs_lin/arduino-1.8.3/hardware/tools/avr/lib/gcc/avr/4.9.2/include/stdbool.h:43:0: note: this is the location of the previous definition
 #define true true
 ^
In file included from /home/tingo/Arduino/libraries/grbl/grbl.h:43:0,
                 from /home/tingo/Arduino/libraries/grbl/examples/grblUpload/grblUpload.ino:27:
/home/tingo/Arduino/libraries/grbl/nuts_bolts.h:56:0: warning: "max" redefined
 #define max(a,b) (((a) > (b)) ? (a) : (b))
 ^
In file included from sketch/grblUpload.ino.cpp:1:0:
/zs/tingo/progs_lin/arduino-1.8.3/hardware/arduino/avr/cores/arduino/Arduino.h:93:0: note: this is the location of the previous definition
 #define max(a,b) ((a)>(b)?(a):(b))
 ^
In file included from /home/tingo/Arduino/libraries/grbl/grbl.h:43:0,
                 from /home/tingo/Arduino/libraries/grbl/examples/grblUpload/grblUpload.ino:27:
/home/tingo/Arduino/libraries/grbl/nuts_bolts.h:57:0: warning: "min" redefined
 #define min(a,b) (((a) < (b)) ? (a) : (b))
 ^
In file included from sketch/grblUpload.ino.cpp:1:0:
/zs/tingo/progs_lin/arduino-1.8.3/hardware/arduino/avr/cores/arduino/Arduino.h:92:0: note: this is the location of the previous definition
 #define min(a,b) ((a)<(b)?(a):(b))
 ^
In file included from /home/tingo/Arduino/libraries/grbl/grbl.h:43:0,
                 from /home/tingo/Arduino/libraries/grbl/examples/grblUpload/grblUpload.ino:27:
/home/tingo/Arduino/libraries/grbl/nuts_bolts.h:61:0: warning: "bit" redefined
 #define bit(n) (1 << n)
 ^
In file included from sketch/grblUpload.ino.cpp:1:0:
/zs/tingo/progs_lin/arduino-1.8.3/hardware/arduino/avr/cores/arduino/Arduino.h:123:0: note: this is the location of the previous definition
 #define bit(b) (1UL << (b))
 ^
Sketch uses 30320 bytes (93%) of program storage space. Maximum is 32256 bytes.
Global variables use 1631 bytes (79%) of dynamic memory, leaving 417 bytes for local variables. Maximum is 2048 bytes.
Low memory available, stability problems may occur.
```
I can also try removing the Arduino IDE completely, if that is needed.

chamnit, 2017-07-11:
@tingox : Well don't know what you to tell you. I just downloaded the most recent version of the Arduino IDE 1.8.3 (Mac OS X) to test if they changed something. They didn't. Importing via 'Add .ZIP Library' and compiling the GrblUpload example sketch with the UNO board works as it should. No errors. 

Something is definitely screwy with your system. Not sure if it's Linux version or something with your install. If it was the Linux version, we would have heard problems back in May when the IDE version was released. So that leaves something with your system.

You can always use the Makefile. It'll make ensure that the Arduino library files are not included. 

Unfortunately I don't have time to help you any further. I have some work-related deadlines I need to be doing.

chamnit, 2017-07-11:
Also make sure you are copying the `grbl` folder INSIDE the downloaded ZIP. Not the unzipped folder. As the wiki instructions explicitly state.

closed 2017-07-11 by chamnit

tingox, 2017-07-11:
Hello,

On Tue, Jul 11, 2017 at 7:01 PM, Sonny Jeon <notifications@github.com>
wrote:

> Also make sure you are copying the grbl folder INSIDE the downloaded ZIP.
> Not the unzipped folder. As the wiki instructions explicitly state.
>
> Well, if I try to select the zip file, I can only the select the zip file
itself (the dialog of the Arduino IDE doesn't allow me to select anything
inside the zip file), and I end up with the error:
Specified folder/zip file does not contain a valid library

if I unzip the zip file first, and then add the grbl folder inside the
grbl-xxx folder, I get this:
java.io.FileNotFoundException:
/home/tingo/dl/grbl/grbl-1.1e.20170114/grbl/grbl (No such file or directory)

No, scratch that, I (finally) figured it out: in the dialog box where you
can select files and folders there is a folders pane on the left side, and
a files pane on the right side.
To select a folder (the 'grbl' folder inside the grbl-xxx folder in this
case) you have to select it in the left pane (folders) - if you double
click on the folder (as you would for a file, you get the error above).
This might be obvious to many (most?) people - it wasn't to me. So I
learned something.

And now the grblUpload sketch compiles without problems (I don't have an
Arduino connected to this machine, so I can' try uploading just now, but I
suspect it will work without problems).
So yes, user error.

Is the Arduino IDE supposed to be able to "look inside" the .zip file when
you add the library? If not, perhaps the documentation could be improved -
mention that one needs to unzip the file before adding (on Linux at least).

Regarding "time to help" - I report problems in projects I use in the hope
that it will help, and that someone will look at the report when that
someone find time to do so. I certainly don't expect anyone to act as a
"help desk" and answer / followup on my reports as if they get paid to do
it.
So please, work on issues as time permits and not more than that.
Have a nice day.
-- 
Regards,
Torfinn Ingolfsen

chamnit, 2017-07-12:
@tingox : The wiki states explicitly [here](https://github.com/gnea/grbl/wiki/Compiling-Grbl#via-the-arduino-ide-all-platforms-recommended-for-all-users) to unzip the downloaded .ZIP file in the very first step. Then it lays out the process for adding the folder in step 3. The error you got when trying to 'Add .ZIP Library' should have been a signal that you were doing something wrong. 

Please understand that Linux installations can and do get really screwy, depending on what a user has installed on their machine and things they altered/hacked in the past that they know about or forgot.  If the 3 things I asked you to check didn't work, which were are all common problems (and easy to search for), then it was likely a problem with your Linux installation. It's impossible to know a user's level of experience with Linux and whether or not they know what they are talking about. Resolving problems like these usually become a massive time sink of going back and forth. When often the solution was an issue that was completely unrelated to Grbl. This is what I'm referring to as not having time to help you with.

michmerr, 2017-11-17:
@chamnit @tingox 

tl;dr: I think that the Arduino IDE adds include lines at the top of the .ino file in some circumstances.

I'm seeing similar issues building the grblUpload.ino. I'm using a cloned repo, and I'm symlinked the grbl directory into my Arduino libraries directory (which works fine.) _But_ compiling _without changing_ anything resulted in issues because of missing definitions. I was able to get past it by changing where grbl.h is included in config.h, but then I read the discussion in this issue.

My grblUpload.ino has a bunch on includes inserted at the top. I didn't do this, and git says they're not in the repo version of the file. All that's left is the Arduino IDE. Did it add those includes as some sort of (faulty) attempt at optimization? Perhaps when adding the library? I just reset the two altered (one by me, one by something else) files to HEAD, verified the extra includes were not in the grblUpload.ino, and recompiled without issue. I suspect that the Ardunio IDE is gaslighting me.

I haven't figured it all out yet, but I thought it worth noting (for anyone searching through issues as I did) that just because you didn't modify the grblUpload.ino file, doesn't mean that something else didn't.

bgrumbine, 2018-06-15:
@michmerr 

I was seeing the same "warning: [...] redefined" messages using with Arduino IDE v1.8.5

Fixed the problem by:
File -> Preferences -> uncheck "Aggressively cache compiled core"

I added a note to: https://github.com/gnea/grbl/wiki/Compiling-Grbl

chamnit, 2018-06-15:
@bgrumbine : Thanks for updating the wiki.

wtarreau, 2019-09-01:
@tingox : thank you very much for the hint about only clicking on the grbl name on the left pane inside the extracted folder. I tried I don't know how many combinations of similar actions in the shitty import box for half an hour, all the time getting "specified folder does not contain a valid library". I even noticed that it would automatically fill-in the directory name and try to open "grbl-master/grbl-master", or "grbl-master/grbl/grbl"... But despite all my attempts I failed to fall into exactly yours which now works. Thanks!

daveapsgithub, 2019-11-23:
There may be more to it.  I struggled with this for about an hour on 2 difference computers.  I read all of the comments above in detail. Although I could bring the library in successfully, I could not compile the grblUpload due to the "warning: "max" redefined" or similar errors. 

Unfortunely I made two changes at the same time. Either one or both contains the secret sauce to solving the problem at least for me.

Make sure that you open the "grblUpload.ino" from "C:\Users\[user]\**Documents**\Arduino\libraries\grbl\examples\grblUpload\grblUpload.ino" and NOT from the directory that you extracted the zip file into.  (eg C:\Users\[user]\**Downloads**\grbl-master\grbl-master\grbl\examples\grblUpload\grblUpload.ino)  This seems to play havoc with the way Arduino IDE deals with search paths but I am unfamilar with the details.

Then, based on the tip from @michmerr (see above), I went into Preferences (never looked in there before) expecting to find "Aggressively cache compiled core" but at least on Windows it was not present.  While in there I turned on Line numbers!    (I hope this is not what fixed it)

It started compiling after that.  

But I find myself with an Arduino Mega (and GRBL is incompatible with it for some crazy reason) so off to Jaycar tomorrow for me to get a Uno.

I hope that helps.

Dave A

3RD-Dimension, 2019-11-23:
Dave, if you have a Mega there is another branch specifically for the Mega version of GRBL.

daveapsgithub, 2019-11-23:
Oh wow! Thanks for the tip! I missed that. Thanks!

oridag, 2021-09-12:
> Then, based on the tip from @michmerr (see above), I went into Preferences (never looked in there before) expecting to find "Aggressively cache compiled core" but at least on Windows it was not present. While in there I turned on Line numbers! (I hope this is not what fixed it)
Boom! That did the trick. How strange...

jayzmiles, 2021-09-17:
> Boom! That did the trick. How strange...

Same thing - this also worked for me on mac

kirkkohler, 2021-12-21:
How strange. Opening the grblUpload.ino file on Windows and in Arduino app > File > preferences clicking "Display line numbers." Worked for me too. I can not upload the file to my Arduino Uno.  Thank you all for continuing to troubleshoot.

drf5n, 2022-04-15:
Those aren't errors, they are warnings.

Since Arduino preprocessing silently adds an `#include <Arduino.h>` to the .ino file if it doesn't find one mentioned, another workaround is to add this clause to grblUpload.ino:

    #if 0
    #include <Arduino.h>
    #endif

Another workaround is to use the https://github.com/gnea/grbl/wiki/Compiling-Grbl#build-as-a-normal-arduino-ide-project method.  It also allows you to maintain and use multiple distinct configurations, such as grbl, grbl-Mega, and grbl-Mega-5X, along with different config.h, defaults.h and cpu_map.h