gnea/grbl #229: Programming lines and arcs

adamonjulian, 2017-07-11:
Hello, I am currently studying the GRBL libraries but I can not understand how to perform the basic routines of straight lines and curves , Since the velocity of the x-axis and the y-axis Changes to be able to make an arc, but in straight lines the speed is constant but can be the same or different, I'm controlling engines step by step, I managed to make straight lines, But I still do not understand how I can vary the speed of the two engines at the same time in a linear code structure as is the C language

adamonjulian, 2017-07-11:
Could you explain how this process is performed or how grbl performs and where these lines are in the code, it is already a little difficult to try to desifrarlo and that in English,  Sorry for the writing was with the help of a translator

chamnit, 2017-07-11:
You are asking a question that requires an in-depth answer that will likely take pages and pages of text and significant time to write. The source code is well commented and organized. I suggest you start by reading it. Motion control along a fixed path with multiple axes is more complicated than most people think.

closed 2017-07-11 by chamnit