gnea/grbl #232: GCode Operational Codes (o codes) Question

CNCQuilter, 2017-07-21:
Are GCode operational codes (O Codes for Subroutines, Loop, Repeat, etc) supported in .grbl particularly when using UGSPlatform?

davidelang, 2017-07-21:
I don't believe so. grbl implements a subset of what linuxCNC supports, and I 
don't believe those are in the subset.

It's basically the minimum needed to support what CAM software actually outputs, 
as opposed to the full power of gcode.

There is a page on the wiki that details exactly what codes are supported.
k

chamnit, 2017-07-21:
@CNCQuilter : No. Macros and subroutine programming are not supported for very good reasons. First, loops are impossible with a streaming interface without a protocol that is able to request specific lines when needed. Second, Grbl has completely maxed out the available flash and RAM on the 328p processor. Third, modern programming languages, like Python, CAM packages, and good GUI programs (bCNC) are much more powerful and can do infinitely more than the very simplistic g-code version of programming. In general, g-code macro and subroutine programming is really only used by professionals who have limited controls at their machines and need to alter a program to save time/money. Hobbyist really don't need this primarily for the 3rd reason.

closed 2017-07-21 by chamnit

CNCQuilter, 2017-07-21:
Sony,

Thank you for your very prompt reply and spot on answer.

For me, the work around is simple - just repeat the block of instructions multiple times.

BTW, I am super pleased with your efforts.

Dave Kender

CNCQuilter, 2017-08-26:
Sonny,

Hello again.

I am using your .grbl firmware in three separate and distinct CNC applications, each with their own dedicated Arduino UNO R3, Protoneer GRBL Compatible CNC Shield, and Universal G-code Sender Platform.

I am working on another project (non-CNC) which uses three-axis stepper motors. The smooth integration of .grbl directed movements would be ideal for the project. However the commands would be event driven rather than canned motion instructions. Are you aware of any instances of .grbl being used in such a manner? If not .grbl, can you recommend another Arduino Stepper Library routine that smoothly and efficiently integrates three-axis stepper motor motion? Since my skills are more along the lines of integration rather than programming, it is far easier for me to adapt the creative work of other’s into my projects.

Thank you,
Dave Kender

Closed #232<https://urldefense.proofpoint.com/v2/url?u=https-3A__github.com_gnea_grbl_issues_232&d=DwMCaQ&c=3buyMx9JlH1z22L_G5pM28wz_Ru6WjhVHwo-vpeS0Gk&r=zO1BysYTt4UFmSFEv9o0SbJim_z9Tb-G4UxWKuM9N7I&m=HaJs_J3mEj_5yh85HcYE702oeUTfUFrBIkD-sgZhdyw&s=gFK91r8u75Pw1ziq-OVeUz39dENhvT6LbEci-MZOc8w&e=>.