gnea/grbl #233: Spindle enable Pin

thester198, 2017-07-24:
I know this is an issue that has been going on for a while but I cant seem to find anyone talking about it.

when first opening the com port for the arduino uno with grbl installed the spindle pin will go high for roughly one second causing the spindle to kick in and if there is an issue and it goes high and low multipliable times in a row, it will trip the GFI i have connected. 

is this a hardware limitation or is there something im missing? if so is there any hardware fixes that anyone knows about?

gerritv, 2017-07-24:
This is due to the boot sequence of the Arduino. It is made worse when you connect to a parallel port breakout board for your drivers because for a short time the drive signal is inverted from inactive.

Opening the com port with trigger a reset of the Arduino, making the problem more visible as well.
The HW solution is to gate the Spindle signal with something else that is inactive during boot/initialization time.

There is some discussion of this in the Grbl Issues https://github.com/grbl/grbl/issues/479

chamnit, 2017-07-24:
@gerritv is correct. It's the Arduino boot loader that does this. There is a FAQ answer related to this [here](https://github.com/gnea/grbl/wiki/Frequently-Asked-Questions#my-cnc-moves-erratically-when-i-boot-up-my-arduino-why-does-it-do-this). I'll update it and add a note for spindle enable.  You can get rid of the boot loader if you use a ISCP programmer or a spare Arduino and just flash Grbl directly to the board. The process is the same, except you just select a different programmer like Arduino as ISP or AVR ISP.

closed 2017-07-24 by chamnit