gnea/grbl #234: Deviation in diagonal cut

freetoair, 2017-07-25:
This error occurs systematically and always in the same place. I tried to reduce speed but do not help. Otherwise G-cod is generated in Sheetcam using Mach3 post processor. In any simulator the cut-out looks great, but when the cuts on the machine always show a deviation at the same place !

 Any help ?
[image]
[image]
[image]
[Triangle.zip](https://github.com/gnea/grbl/files/1174042/Triangle.zip)

chamnit, 2017-07-25:
@freetoair : A quick test of the gcode you attached showed no errors. There is no reason to believe there is a Grbl-specifc problem. Otherwise, we'd literally have thousands of reports of position issues  for a simple job like this one. 

From the picture, it looks like you have a mechanical problem, where you may have some binding, resulting in your steppers losing position. Grbl should be reporting the correct positions throughout the job, because it tracks position by directly counting the step pulses sent to your stepper drivers. If Grbl is indeed reporting the correct position, then it's something between your stepper drivers, stepper motors, to your axes and tool.

closed 2017-07-25 by chamnit

jahnj0584, 2017-07-25:
Looks like electrical noise. Make sure everything is shielded, espically
since you are using plasma/lazer/whatever

electrokean, 2017-07-26:
I agree with @chamnit that this looks like a mechanical/binding issue.  Can you try cutting the same shape using a different home/starting position.  i.e. further in from the machine edge that show the problem.
An electrical noise issue is unlikely to manifest in this way, nor would it be easily repeatable.

freetoair, 2017-07-26:
Yes, I did not even suspect that this was a problem in the grbl. However, as far as the mechanical problem is concerned, I tried to move the starting position and everything is the same. This means it's not a mechanical problem. The next is that I have separated all the galvanic connections between the Arduino, which is with the 16U2 chip, and the other components. The computer is Lenovo IdeaStick and the software for sending the code is Candela, which is managed through Team Viewer. Limit switches are connected via opto-couplers. Arduino never locked up at the start of the plasma generator. Connections between Arduino and drivers with TB6560 are 5 cm (2 inches) long. Of course I tried with other forms, but the place where deviation appears is completely undefined, and also not always in the same place on the same form. I state all these facts without some sort of order, as I thought about it and rejected it one by one.

chamnit, 2017-07-26:
@freetoair : After a shift, did you check whether or not Grbl is reporting the correct position, according to the gcode? Again, if it is, then it's not Grbl. Grbl tracks position at an interrupt level via counting the step pulses it sends to the drivers. So, if its right, then Grbl is doing its job. 

If you don't believe me, you can try programming another Arduino to count the pulses and track position externally by tying directly into the step pulse and direction pins. This is a test I do frequently. Here's a rudimentary sketch that will track one axis. 
[Pulse_counter.txt](https://github.com/gnea/grbl/files/1177579/Pulse_counter.txt)

109JB, 2017-07-26:
It looks to me like the even after the diagonal cut the end of the cut winds up at the correct position.  If this is true then the only explanation is that the machine is flexing or binding without actually losing steps.  If it were lost steps then the end of the cut would be off because lost steps cannot be compensated for in an open-loop control such as Grbl. As chamnit mentioned the fist thing to check is if the machine still knows its position correctly after one of these cuts.  You can do this by having a reference location that you can return to after a run.  If the machine returns correctly to the reference position then it isn't lost steps.

freetoair, 2017-07-26:
The first thing I wanted to do is to thank you for your time for my problem. In the meantime, I tried to cut out some other forms that would help me to understand what is happening. When I cut this four form shown that the error does not occur only on the diagonal cuts. All these four forms were cut out one at a time, in different places on the board that differed from more than the width of the form. Of course, with the same G code
[image]
[Mreza.zip](https://github.com/gnea/grbl/files/1178280/Mreza.zip)

. The only form that is the first on the left is good, and it is cut out the last one.
I'm turning on now Arduino to counting the impulses with the sketch you sent me.

freetoair, 2017-07-26:
It's very late for me now to start a plasma cutter, so I'm going to try with the pens to check if it's getting back to the beginning, as suggested 109JB.

neilferreri, 2017-07-26:
@freetoair I agree with the others that this looks mechanical... Binding, skipping steps in your Y-axis (if parts are oriented as they were on machine bed). There could be a number of things that cause this depending on your machine. If your machine is belt driven, check that your drive gears are not loose on the stepper shaft. A very common, often overlooked issue.

zeevy, 2017-07-27:
I had a similar problem, in my case the motors are not getting enough current. The jumper settings in driver board are wrong. Make sure you have the correct settings for the current and micro stepping.

freetoair, 2017-07-27:
The problem is solved. Now everything works fine. And of course the problem was not in grbl, it was a mechanical cause. Thank you all for keeping me focused on the mechanical problem. Namely, the axis of the step-motor for the X axis was not tightened enough. And that's all, I hope this my experience to help someone in the future. Thank you once again to everyone who participated in this discussion. 

regards, freetoair

X3msnake, 2017-07-27:
Cool, great that it is solved ;)

Thanks for sharing