gnea/grbl #235: laser engraving: M4 vs M3

kees1948, 2017-07-26:
Hello, I'm playing with laser engraving. I'm using inkscape/bCNC/grbl1.1 to control the process and the hardware.
I'm experimenting with engraving text, so in inkscape I defined 3 textlines with different sizes.
What I observe is that with the M3 start codes everything is done but heavily. With the M4 start code
the engraving is _much_ nicer but some start moments are missed in the result.
Is there anything that I can test further why this happens?  Attached is a jpeg where the differences are marked, it is the smallest line from the three, the height of the 'D' is about 8 mm. With the M4 code it appears very nice but the laser is switched on too late at some points.
The g-code file is about 250kB and it is identical for both tests except the M3/M4 start code. If someone wants it I'm happy to send it.

[image]

chamnit, 2017-07-27:
Grbl adjusts the PWM output for the laser based on current speed. The circled parts of the image are where Grbl accelerates. So M3 burns a little extra while M4 gets adjusted  My guess is that your laser doesn't either turn on when the PWM is too low or your laser burns the material differently at low PWM. You can alter the minimum PWM level in the config.h file. Grbl treats it as linear from min rpm to max rpm.

kees1948, 2017-07-27:
I give it a try. What was visible was that the laser was delayed. For instance the 'dots' on the 'i' where not laser-ed the first turn but only on the second turn.

chamnit, 2017-07-27:
@kees1948 : Can you provide a few things? I need your $$ Grbl settings, your $I build info string, and a copy of your g-code program.

kees1948, 2017-07-27:
Via the forum or in email?

chamnit, 2017-07-27:
Here. Just alter the file type suffix of your gcode `.nc` file to a text file`.txt. You can then drag and drop the `.txt` file into the response.

kees1948, 2017-07-27:
[image]

[M3_output_0008.gcode.txt](https://github.com/gnea/grbl/files/1181275/M3_output_0008.gcode.txt)
[M4_output_0007.gcode.txt](https://github.com/gnea/grbl/files/1181277/M4_output_0007.gcode.txt)

Here you are

chamnit, 2017-07-27:
I still need your build info $I strings.

kees1948, 2017-07-27:
ok
$I
[VER:1.1f.20170531:]
[OPT:V,15,128]
ok
$G
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
ok

mayhem2408, 2017-07-27:
@kees1948 @chamnit I use M4 mode all the time. It works very well but if you have a machine with slow acceleration, you will see this problem. The simplest explanation is if you feed at 3000mm/min at 100% laser power, you will burn a line, however if you reduce everything to 10%, so feed at 300mm/min at 10% laser power,  you will not burn anything because the laser does not have enough power to significantly heat the material. M4 works really well at preventing overburns, but with slow accelerating rigs, you may get unburnt areas. 
@chamnit Other controllers get around this problem by using ppm, or pulses per mm.  This means if you are set to 60ppm, regardless of the speed you are traveling, the laser will only fire 60 times in 1 mm. Might be something to look into in a future release.

kees1948, 2017-07-27:
[image]

These were the settings in inkscape for the output file generator. I don quite understand the implication of 'slow accelerating' bCNC sends the G-code, grbl interprets it, what can cause this effect? You mean the controlling hardware on the board? Thanks anyway for looking into it.

chamnit, 2017-07-27:
@mayhem2408 : Understood. Unfortunately the limitations of the 328p processor doesn't allow for PPM. You need a spare timer, which I don't have.  

However, Grbl's M4 dynamic mode basically does something similar to constant PPM. It ensures the amount of power per mm is constant instead. This is done by varying the PWM width. If the laser responds linearly with RPM, then it should work fine. But as you said, most lasers aren't. 

As an interim solution, I've installed a nonlinear spindle PWM capability for specific use cases like lasers. There is also some instructions on how to generate the constants to input into Grbl for it work. All free and solved via a web repl of Python.

langwadt, 2017-07-28:
if limited to drivers with fast step inputs couldn't you hijack timer0?

mayhem2408, 2017-07-28:
@chamnit There is more to it than the laser power being linear. I have a custom driver with produces as close to a linear output within as we could get. My diode laser rig is rated at 5W and when measured it produced 4.7W output. When I run it at 10%, I get 0.46W and 50% got 2.33W, so it's pretty close. The other factor to take into account here is the materials ability to absorb the lasers energy and dissipate it. With 5W hitting a piece of wood for 1s, the wood can not dissipate the heat quick enough and will burn. If you pulse that 5W at only 10% duty for 10 seconds, the wood surface will heat up but before it gets hot enough, the laser turns off and the built up heat dissipates and when the laser turns back on, the surface temperature has to heat up again only to cool down again. 

I have my own tables of what power curves work with different materials. Some materials do nothing up until 30% power, some show signs as low as 5%. 

One way to over come this, which I have done on one of my Marlin firmware lasers is to have a minimum laser value, so a movement scale of 0-100% might power the laser from 25-100%, This will eliminate the low power no burn situation. The problem is you can't really hardwire a minimum as it is different for different materials.

kees1948, 2017-07-28:
'How' do you measure the laser output power?

chamnit, 2017-07-28:
Good question. I'm not sure. Spindle RPM is relatively easy. Lasers tend to burn materials at different rates, so a pure watt measurement (Volts*current_drawn) may not be accurate. Also the laser could and will likely respond nonlinearly to power input, where excess energy is lost to heat.

However, it'd be a good start to measure power draw. You can set different `S` in M3 constant power mode and quickly measure the volts and amps. 

I bet there are more accurate ways to measure laser output, but I'm not too versed on laser cutters. So, if anyone else knows of an alternate method, hopefully they'll post it here. I'm curious too.

kees1948, 2017-07-28:
Would a controller with a Mega2560 be more suitable here?

mayhem2408, 2017-07-29:
@kees1948 I have a friend that works for a company with a sensor that can measure a lasers output energy. He said it is something like this. https://www.coherent.com/measurement-control/measurement/sensors-laser-power/air-cooled-thermopile-sensors-to-150w-rohs He said it is very expensive. In excess of $1000. We used it to measure the lasers output and calibrated the driver for a near perfect linear output based on a 16kHz PWM input.

chamnit, 2017-08-20:
Closing due to interim solution with a nonlinear spindle PWM model. Measuring and calibrating dynamic power for engraving is clearly tricky. I'll make a note for future versions to try to find a firmware solution for this.

closed 2017-08-20 by chamnit

Fusseldieb, 2020-03-02:
> You can alter the minimum PWM level in the config.h file. Grbl treats it as linear from min rpm to max rpm.

I had success altering the `cpu_map.h`. There's an option called `SPINDLE_PWM_MIN_VALUE`. Sometimes 0 (which is the default) is too weak for lasering, resulting M4 to go too low and not even engraving, so I set it to 10 (May vary greatly, on another machine I set it to 60. Test it thoroughly!). 

With MIN 10 and MAX 1024 (on mine!) it works well. Combining it with M4 S128 (50% duty cicle, $30 set to 255), it engraves just nicely without burning or weak spots.