gnea/grbl #236: Cant turn the spindle on and off

aesarius, 2017-07-27:
Hello,
I have just upgraded form .9 to 1.1 and I cannot turn on my spindle using M3 S1000 nothing happens?
I have set up min and max speed. Is there something else to do except just flashing the new 1.1, set up and go?.. Using Uno/Cnc Shield.
Cheers.

chamnit, 2017-07-27:
@aesarius : Do you have laser mode ($32) enabled?

chamnit, 2017-07-27:
@aesarius : Also, is your CNC shield pin configuration compatible with v1.1 and default 0.9 builds? Spindle enable/PWM is on pin 11, not pin 12, which is what most CNC shield clones are configured for. You can either alter your CNC shield or disable VARIABLE_SPINDLE in config.h (recompile and flash), which will move spindle enable back to pin 12..

closed 2017-07-27 by chamnit

aesarius, 2017-07-27:
Hey chamnit,
Thanks so much for your reply. Sorry kinda new to the whole grbl thingy :D
laser mode is set to $32=0 so disabled yes.
Was using SpnEn on .9 and everything was working ok.
I have changed config.h to use #define USE_SPINDLE_DIR_AS_ENABLE_PIN and the spindle is starting now but when I start it up the spindle will come alive for a few seconds, is this normal?
I might take your advice and disable VARIABLE_SPINDLE but if i can I would like to get the full Grbl experience ;) Cheers!

chamnit, 2017-07-27:
See [this FAQ](https://github.com/gnea/grbl/wiki/Frequently-Asked-Questions#my-spindle-turns-on-when-i-boot-up-my-arduino-for-a-second-or-two-why-does-it-do-this).

aesarius, 2017-07-27:
I will thanks again!