gnea/grbl #237: CAMWorks GRBL postprocessor

johnlife, 2017-07-29:
Just in case someone will use GRBL with CAM module for SolidWorks that will become official SolidWorks' CAM module next year (http://blogs.solidworks.com/solidworksblog/2017/02/introducing-solidworks-cam-smart-manufacturing-ecosystem.html). And the best CAM embedded in SolidWorks tested so far. Here is the simplest possible postprocessor for our controller. 

It supports linear and arc moves, variable spindle speed and no tool length compensation. 
Just unzip it somewhere on hard drive and point CamWorks to it
 Machine->Edit definition->Post processor tab->Browse->Select
Make sure you've selected it right:
[image]

[grbl_post.zip](https://github.com/gnea/grbl/files/1184860/grbl_post.zip)

Disclaimer: I'm really a newbie at G-code and/or CamWorks, so THE FILE IS PROVIDED AS_IS with no warranties. Try it first in the air before you crash your machine.

X3msnake, 2017-07-29:
> introducing-solidworks-cam-smart-manufacturing-ecosystem.html). And the

cool, thanks for sharing,

But you should put this on a github!

closed 2017-08-01 by chamnit

nyakacs, 2018-01-29:
Hi Johnlife,

It did happen and Solidworks 2018 has built in CAM. Since I have Both Solidworks and a GRBL based small milling machine I am very interested using the post processor you started. Could you point me to any documentation you used to get so far? Solidworks support is not interested helping at all. They only work with big machines....

jahnj0584, 2018-01-30:
I have a custom post processor for grbl made. Currently sick but when I'm
on the computer tomorrow i can share.

X3msnake, 2018-01-30:
Share it in a Github if you can ;)

waqaskhan92, 2018-03-07:
i cannot get access to the post processors in Cam works 2018 only i have access to milling machine will you please help me?

X3msnake, 2018-03-07:
Use fusion 360

SAJamesH, 2018-03-12:
I have been trying to get Solidworks CAM (Camworks) up and running with GRBL and the Universal GCode Sender and an XYZ Carve (similar to an XCarve) with limited success.
Straight line cuts are fine, it doesn't seem to like arcs at all. I've tried adjusting a lot of settings, and used the POST shared here without much success. Mostly it gives error 33 but a few others also. 
Does anyone know how to get it all working? It seems to be something between the arcs solidworks generates and the precision GRBL expects.
Thank you for any help.

swarfer, 2018-03-12:
you will need to output 5 digits of decimal precision if in metric mode.  don't know about inches but maybe 6?
thing is that many of these CAM systems output really small arcs, around 0.1mm arc length, and at that size the precision of the end points is crucial.

Make sure the UGS is  NOT truncating the decimal precision!  UGS has a bunch of background settings that can break perfectly good Gcode.

can you increase the minimum chord length?  about 0.5mm should do it from my experience with the Fusion360 post.

jahnj0584, 2018-03-12:
I post out three decimal places in metric without issue.

SAJamesH, 2018-03-12:
Thanks for the help.
It seems Camworks is only outputting 3 digits, I need to work through how to increase that. I have dug through every setting I can within Solidworks/Camwors, I haven't found a chord length setting, it may be the 'spline precision' default =0.03, I'll try changing that to 0.5. Failing that it needs a custom post through the Universal Post Generator
Otherwise can you tell me what the UGS background settings are and how to change them?
$12 (recommendation 0.002) Seems to be a check more than a functional command.

SAJamesH, 2018-03-13:
I found the GRBL settings....
Truncate decimal digits - 4
Convert arcs to lines (G2/G3 ->G1)- N
Small arc threshold (G2/G3 ->G1)mm - 2
Small arc segment length mm - 1.3

Can anyone suggest optimum settings?

I still think the base problem is Camworks only seems to be outputting 3 decimals in the first place.

swarfer, 2018-03-13:
maybe this can help you modify the post
https://www.youtube.com/watch?v=vXe0s5IbpC4

piflixe, 2019-04-15:
dear @johnlife thank you so much for sharing your GRBL postprocessor. I would like to use it with [Smoothieware](http://smoothieware.org/) and I think I have to edit it a little bit for that. Would you mind also sharing the sourcefiles?

piflixe, 2019-04-15:
to anyone who is still searching on this: http://www.restrictedayerspace.net/cnc/grbl-postprocessor-for-solidworks/

msbealo, 2021-05-06:
Hello,

If anyone is interested in using Solidworks CAM or CAMWorks then I've put together a post processor for GRBL machines. It's hosted at [https://github.com/msbealo/GRBL-Post](https://github.com/msbealo/GRBL-Post)

It's still under testing and development but I think it looks about right. I've been testing it on my controller but I've not made any cuts yet as my machine is in a rebuild. Please let me know if you would like any modications.

P.S. Please use caution.

SAJamesH, 2021-11-30:
Hi
I haven't watched the 1 hour UPG tutorial as yet...
Can anyone tell me if and where there is a feature in UPG to not used GCODE that GRBL can't accept?
Thanks

SAJamesH, 2022-01-31:
In case anyone is interested, the root of the problem, for me at least, seems to be that GRBL gets confused by cutter compensation - G41 and G42 - on small arcs.

This can be switched off in CAMWorks - in Operation Parameters, NC, set 'CNC Compensation' to off.

This took a lot of time to figure out, I have not worked out how to disable G41, G42 by editing a custom post or any means other than turning off in CAMWorks

There could be some other factors, I believe I have typical GRBL and sender settings but all the public Post Processors I've tried have the same issue so maybe its specific to my setup.

It seems to be caused mostly by arc leadins, but bobs up from time to time with small radii.

GrahamJB, 2022-01-31:
As far as I am aware Grbl does not support G41 or G42 They should be rejected by Grbl as an error 20.

 

G40 is supported but it does nothing, It’s just there to prevent errors if the Gcode specifically states Cutter compensation is off.

SAJamesH, 2022-01-31:
It could be my GRBL build needs an update, currently it doesn't error 20 G41 or G42.

GrahamJB, 2022-01-31:
I’m on 1.1f

 

My answer was based on the github documentation, not testing it.

SAJamesH, 2022-01-31:
Also 1.1f.

From the history I can see that GRBL is sensitive to number of decimal places and arc dimensions  when running arcs, error 33 can get triggered if everything doesn't match up.

SAJamesH, 2022-02-06:
I found this in a guide to MACH4

"Cutter comp should be enabled on a linear lead-in move; an error will be produced if cutter comp is
enabled in a block with an arc. If there is a value other than zero in the offset then this lead-in move
may not be parallel to the programmed path, see figure 28, 34 and 35. The end point of the start
block is the point at a 90° angle to the movement in the next block and at the offset distance. See
figure 30 for examples. This linear move must be longer than the offset amount, if it is not there will
be an error. Also, if there are any segments of the tool path shorter than the offset amount a gouge is
likely, see figure 31. The tool path display in Mach will display the actual tool path with comp, check
to make sure there are no gouges or abnormalities before running a part."
https://machmotion.com/documentation/Software/Mach4/Mach4-G-and-M-Code-Reference-Manual.pdf

This is exactly the problem I've been having, mainly on lead-ins with arcs but also other small arcs, I think.
It would be great if I could figure out how to disable cutter comp via the post-processor.
I'm guessing the CAM software most people use to write GCODE for GRBL doesn't use cutter compensation, and the senders don't have the facility to use it,  so few people see this problem.