gnea/grbl #238: Override Feed has some problems?

viewsat, 2017-07-31:
The following two pictures of my Grbl settings, when x or y or z [accel] is set to 150 ($ 120 or $ 121 or 122 = 150). Send gcode F = 100, Override Feed down to 10, There will be an error.

I use the G-code:
G21
F100
G1X200
G1Y150
G1Z50
G1X0
G1Y0
G1Z0
G1X200
G1Y150
G1Z50
G1X0
G1Y0
G1Z0
M30

[image]
[image]

chamnit, 2017-07-31:
You'll need to provide more details than that. What kind of error? Overrides shouldn't have any error associated with their commands. How are you sending overrides and are they actual Grbl override commands or a GUI based one that just alters the feed rate word F.

viewsat, 2017-07-31:
I will record a video to see this error.

viewsat, 2017-07-31:
@chamnit  Please watch my recorded video to understand this error.

see vedio>> https://youtu.be/qFnzyvqSxfk

chamnit, 2017-07-31:
Thanks for the video. That explains your problem perfectly. Could you provide your $I build info output? I need this to establish which version and what options you are running with so I can reproduce the problem. I'm guessing that you have AMASS disabled because you shouldn't be able to run this slow of rates with such a high step/mm setting at 1066 steps/mm.

chamnit, 2017-07-31:
@viewsat : Nevermind. I found your version in your video and successfully reproduced the issue. In the short time I probed the issue, I noticed that it's tied to your acceleration setting for some reason. `$120=150.1`, `$120=149.9`, and pretty much any other value works fine, while `$120=150` causes the lockup issue. It also works fine with `$120=150` if you avoid 40%. I set it at 98% overrides and successfully incremented down to 10% and back to 100% with no problems Can you verify this on your end? 

I have a pretty good idea where this problem might be coming from and will be elevating this to a critical bug. It seems to effect all versions of Grbl v1.1. However, it seems to be specific to a certain rare set of circumstances based on this testing and the lack of other reports like this.

chamnit, 2017-07-31:
@viewsat : I think I located the problem and pushed the fix. Please test this thoroughly and let me know how it goes. Thanks.

viewsat, 2017-08-01:
@chamnit  I tested v1.1f (2017-07-31) Release, it seems that there are some problems, please wait for my test results.

chamnit, 2017-08-01:
@viewsat : Can you be a little more specific and provide a point case? The change was very minor. While there is a chance that the hot fix did not capture all scenarios due to numerical roundoff, it seemed to fix your particular issue in my testing.

chamnit, 2017-08-01:
@viewsat : Been thinking about the problem a little more. I think I came up with a more robust branch calculation. It should completely eliminate the issue. Please try this and let me know how it works.

Replace the entire RAMP_DECEL_OVERRIDE case statement starting on line 807 of stepper.c with this instead:
```
        case RAMP_DECEL_OVERRIDE:
          speed_var = pl_block->acceleration*time_var;
          if (prep.current_speed-prep.maximum_speed <= speed_var) {
            // Cruise or cruise-deceleration types only for deceleration override.
            mm_remaining = prep.accelerate_until;
            time_var = 2.0*(pl_block->millimeters-mm_remaining)/(prep.current_speed+prep.maximum_speed);
            prep.ramp_type = RAMP_CRUISE;
            prep.current_speed = prep.maximum_speed;
          } else { // Mid-deceleration override ramp.
            mm_remaining -= time_var*(prep.current_speed - 0.5*speed_var);
            prep.current_speed -= speed_var;
          }
          break;
```

viewsat, 2017-08-02:
@chamnit I tested the press-2-3.nc this gcode and Overrides Feed = 10% Rapaid = 100%, and when I executed the 283 line G1X33.01 Z-0.583 after completing the next 284 line G0 Z5, then the speed of G0 was wrong And become very fast, more than Grbl setting of the max rate, so that stepper motor is locked can not correct action.

I tested press-2-3.nc G0 speed wrong place there are many, 104 lines, 194 lines, 284 lines, 374 lines ... etc. When there is G0 Z5 line G0 speed is wrong. 

I use v1.1f (2017-07-31) Release & Grbl-Panel 1.0.9.15 to test.
test gcode >> [press2-3.zip](https://github.com/gnea/grbl/files/1192415/press2-3.zip)

[image]

chamnit, 2017-08-02:
@viewsat : Are you absolutely certain that there is a problem with G0? Did you measure the G0 rate with an oscilloscope? You are going from a 10mm/min rate to a 800 mm/min rate instantly. It can seem like a sudden jolt.

Also, this seems like a separate problem from the one we have been working on. Did you verify that the 7/31 build and today's 8/1 push work as intended in removing your prior problem?

viewsat, 2017-08-02:
I tested the 7/31 and 1/8 versions that had no previous problems.

But when I was in the test when G0 has an incorrect action, I do not know whether the G0 speed or acceleration error, this is a new problem.

chamnit, 2017-08-02:
Please show a video with an actual measurement.

closed 2017-08-02 by chamnit

viewsat, 2017-08-03:
@chamnit  I checked my CNC machine and found that my Z-axis screw was missing lubricant.
After I add the oil, the G0 Z-axis speed now runs correctly without problems.