gnea/grbl #239: Which limit switch is tripping

komradebob, 2017-07-31:
Would it be possible to modify the alarm that is tripped when a limit switch is reached to tell the operator which limit switch was hit? 

Today it looks like:

ALARM: Hard limit
[Reset to continue]

I would propose something like:

ALARM: Hard limit - X axis
[Reset to continue]

gerritv, 2017-07-31:
Pn:XYZPDHRS indicates which input pins Grbl has detected as 'triggered'. 

Part of the response to '?' request from the GUI. Some GUI's display this, e.g. GrblPanel latches an indicator for the various pins.

chamnit, 2017-07-31:
@komradebob : If the limit switch is still not engaged when a hard limit is triggered (often happens when noise is present), there is no way to tell which one did*. The AVR processor only provides which port experienced a pin change. It does not specify or record which pin on the port triggered. 

* If by chance the limit pins were all placed on different ports AND no other pin with an external interrupt is used on the same port, then you could tell which pin triggered. You can only do that on something like an Arduino Mega2560, where there are enough pins and ports to do something like that.

closed 2017-07-31 by chamnit