gnea/grbl #240: G28 Homing only Z

yoshinator13, 2017-08-01:
On my CNC, I have a tool that impacts the work surface instead of drilling into it. As a result, my Z axis occasionally becomes uncalibrated. I could home after every impact, but it would increase the time significantly. I did some research, and I believe the G28 command is typically used for homing only the Z axis. This would significantly reduce time.

chamnit, 2017-08-01:
G28 has multiple definitions, sometimes homing and other times moving to a pre-defined position. Grbl defines it as the latter. Not as homing. It follows the LinuxCNC standard, as clearly stated in the wiki. 

I would invest time into preventing your machine from getting uncalibrated instead. It'll save you a lot of time and energy by doing so. If you can't or won't, you can always use G38.x probing cycles and set a tool length offset by the uncalibrated distance.

chamnit, 2017-08-01:
Also, look in config.h for the line `#define HOMING_SINGLE_AXIS_COMMANDS`. I installed this relatively recently, but have not seen anyone use it yet. But, I would still invest time in ensuring your machine stay calibrated from impacts with a better drive system.

closed 2017-08-01 by chamnit

langwadt, 2017-08-01:
connect the z limit switch to the probe input and do a g38.2...  G10... instead?

yoshinator13, 2017-08-02:
> Also, look in config.h for the line #define HOMING_SINGLE_AXIS_COMMANDS. I installed this relatively recently, but have not seen anyone use it yet. But, I would still invest time in ensuring your machine stay calibrated from impacts with a better drive system.

@chamnit This looks like it will work perfectly. I was did not see this in the config file last time I looked.