gnea/grbl #241: Safety Door & Parking

neilferreri, 2017-08-02:
I think I'm missing something somewhere.  I have an old serial machine that I have converted to use grbl.  It has a safety door that I'd love to use with my current setup.  Right now just testing on an Arduino Uno, using the latest release, I modified config.h to ENABLE_SAFETY_DOOR_INPUT_PIN and PARKING_ENABLE, but I am not seeing what I expected.  The feed hold pin on the Uno works as it did before.  When I ground it, I enter a HOLD : 0 state and I have to cycle start to resume. There is no Z-retraction for the parking (I left the other parameters with the default settings).
My machine space is in the negative and HOMING_FORCE_SET_ORIGIN is not enabled. Like I said, I am just testing on an Uno with a momentary switch that will ground pin A1.
Thanks

chamnit, 2017-08-02:
@neilferreri : The parking motion will enact when there is enough z-clearance to do so. I think it's when Z < -5mm. At Z < -10mm, it'll perform the pull-out motion.

closed 2017-08-02 by chamnit

neilferreri, 2017-08-02:
@chamnit This is machine Z, right? I've had my machine Z everywhere from -20 to -100 and I'm still not getting any parking action.

chamnit, 2017-08-02:
Are you editing the config.h file in the Arduino library?

neilferreri, 2017-08-02:
@chamnit NOW I am...thanks!

TekMason, 2020-03-11:
It was not explicitly said in this thread of what settings to use.  For me it was to uncomment:
```
#define PARKING_ENABLE  // Default disabled. Uncomment to enable
#define ENABLE_SAFETY_DOOR_INPUT_PIN // Default disabled. Uncomment to enable.
```
When the feed hold is put to ground the spindle will raise up, stop the spindle and then go up to z=-5.0

TekMason