gnea/grbl #243: Interactive stream script

fragmuffin, 2017-08-06:
regarding [this code comment](https://github.com/gnea/grbl/blob/master/doc/script/stream.py#L20-L21):

I've written an interactive python script using `curses`.
https://github.com/fragmuffin/pygcode/blob/develop/scripts/grbl-stream

**Jogging**
It starts off in a _jogging_ state with keys:

    ; Jogging Keys                                                                
    ;    - Arrow Keys: X/Y axes                                                   
    ;    - PgUp/PgDn: Z axis                                                      
    ;    - []: increase/decrease jog distance                                     
    ;    - Space: zero work offsets (all axes)                                    
    ;    - x,y,z: zero individual axis                                            
    ;    - Enter: start streaming gcode                                           

[image]

Machine's mode is changed to incremental `G91` if jogging is requested, then reverted back to the initial mode when finished.

**Streaming**

When user presses `[enter]` gcode is streamed

[image]

in both _jogging_ and _streaming_ states, gcode on screen indicates...

    G00 Z2             > ok               gcode was sent, and GRBL device has responded with 'ok' 
    (comment)
    G00 X10            >                  line has been sent, but no response received
    G00 Y-10                             line has not  been sent yet

script continues streaming / checking status until:

- all lines have been sent
- all `ok` responses have been received
- machine's state is `Idle`

**Usage**

    usage: grbl-stream [-h] [--quiet] [-d DEVICE] [-b BAUDRATE] [--disable-status]
                       [--poll-interval POLL_INTERVAL] [--buffer-size BUFFER_SIZE]
                       [--serial-log LOG_FILE] [--pending-count PENDING_COUNT]
                       [--keep-open]
                       infile

    GRBL gcode streamer for CNC machine. Assist jogging to position, then stream
    gcode via serial.

    positional arguments:
      infile                gcode file to stream

    optional arguments:
      -h, --help            show this help message and exit
      --quiet, -q           if quiet, help messages won't be printed

    Serial Connectivity:
      -d DEVICE, --device DEVICE
                            serial device GRBL is connected to, can use direct
                            device name (eg: /dev/ttyACM0) or the Arduino's serial
                            number (eg: 55639303235351C071B0)
      -b BAUDRATE, --baudrate BAUDRATE
                            serial baud rate (default: 115200)

    Debug Parameters:
      --disable-status      disables '?' being sent every '--poll-interval'
      --poll-interval POLL_INTERVAL
                            frequency of sending '?' for status report when status
                            is enabled
      --buffer-size BUFFER_SIZE
                            GRBL internal serial buffer size
      --serial-log LOG_FILE
                            if given, data read from, and written to serial port
                            is logged here (note: \r and \n characters are escaped
                            for debugging purposes)
      --pending-count PENDING_COUNT
                            number of gcode lines to display ahead of them being
                            sent
      --keep-open, -o       if set, window won't close when job is done

**Limitations**

- It's a bit buggy at the moment (I only got it working earlier today)
- won't work on Windows as is (there's a port called `UniCurses` which may work)
- requires the `pygcode` lib

Would you be interested in this being an addition to this project?

chamnit, 2017-08-06:
I would for sure. However, it would likely need to be a linked project, rather than part of the main codebase. This is primarily due to time and effort to ensure things are working and supported well. For the most part, the Grbl streaming script is intentionally simple so there is something to test with when more complex UIs seem to fail. 

That said, I've always wanted to expand the streaming script to something simple like your text-based GUI. In reality, you don't need it to do much more than that for 99% of tasks. Plus, it would likely run great on a small hobby LCD screen on an RPi Zero and a small keyboard. Please keep us updated on your progress.

fragmuffin, 2017-08-07:
> "it would likely need to be a linked project"

I agree, as a single script of 1000 lines is a bit too long for my liking.
I'll split it out into it's own project, and I'll post a link here when it's done.

> "For the most part, the Grbl streaming script is intentionally simple"

I agree... that script is how I learnt to buffer the bytes being streamed over serial, whereas this one is far too difficult to learn from (as a newbie)

> "Plus, it would likely run great on a small hobby LCD screen on an RPi Zero and a small keyboard"

Brilliant idea!, I hadn't thought of that :+1:

fragmuffin, 2017-08-08:
I've created and started the repo here: https://github.com/fragmuffin/grbl-stream

Also, there's another noteworthy contribution to python (that you're probably aware of) [PyGRBL](https://github.com/ajmendez/PyGRBL).

Although my focus for `grbl-stream` is different, so there's room for both in this wonderful world of open software.

I'll re-post here when I release `grbl-stream` on PyPi

fragmuffin, 2017-08-15:
OK!, I've released [grbl-stream](https://github.com/fragmuffin/grbl-stream) that can be used now.

I've pushed v0.1.0 to master (as a _pre-alpha_), but I haven't released it to `pip` just yet (reasons listed in the above link).

It has a few perks I haven't seen in any other grbl streamer, such as:

* always displaying the `ok` or `error` inline with the gcode being streamed
* splitting gcodes and sending them as separate lines (optional)

All things considered, my adventures into the CNC world have largely been a distraction from my main task, so I have to stop development on these scripts for a short while... If anyone shows interest in using them, then I'll gladly pick them back up and polish them up to be a bit more shiny.

@chamnit : if you have any inclination to make this a feature streamer for `grbl`, please let me know and I'll put a high priority on making it much more _portable_ (ie: windows), _reliable_, and _robust_ so there's no reason for users to be discouraged.

chamnit, 2017-08-20:
@fragmuffin : Sorry about the late reply. Been busy. I'd love to make this a featured steamer but it would require some dedication to keeping it maintain by yourself or someone else. UIs die when they are no longer maintained. 

However, I do have some interesting questions I'd like to pose to you offline. Please email me (see my profile).

fragmuffin, 2017-09-03:
(whisper whisper)