gnea/grbl #244: Pins options + Spindle Wiring

jahnj0584, 2017-08-06:
In a week, I'll be making my electronics box and setting up the new machine!
1)Does anyone know where to wire the PWM and Spin Ena pins on a HuanYang VFD? What is the voltage on the PWM anyways? 5V? 

2) I'd like 2 types of coolant on my relay, one is a vacuum (simply switches a power bar from NO to closed) and the other will be a compressor line for milling metals. This may be easier to just do manually, honestly. just asking. 

3) Can I have the other relay control the stepper motors PSU? It'd be cool to have the whole box basically shut itself off via the arduino. Any pin option for that? If not, I'll just use a on/off toggle.

4)How does the feed hold pin work? I'd like to add my E-stop button, but without just losing power to the drives and my work position. The switch has one NO and one NC circuit, so if there was some way I can have it Feed hold when I engage the switch, then reset when I pop the switch out? Is that possible or will it just constantly reset itself as long as the circuit is complete. 

Thanks!

(I'll also be posting this elsewhere but ask that they respond here)

KoljaWindeler, 2017-08-06:
Hmm I always thought this place would be used to post bugs and not to ask generic questions that you can Google yourself ..

jahnj0584, 2017-08-06:
Feel free to link me to an answer then, because I clearly haven't seen them yet.

gerritv, 2017-08-06:
The pins and their descriptions are on the wiki.  PWM is also described.

E Stop is for emergency, wire it to Reset. You want everything to stop, you will of course lose position as well.

Try a CNC router forum for generic questions.

jahnj0584, 2017-08-06:
Either answer me directly or with a link. No need to be rude about it. These questions are not easily found online

chamnit, 2017-08-06:
Just as an FYI, there is no general Grbl forum anymore. It used to be the shapeoko forum but that has since been superceded by Carbide3d after they merged. Most Grbl OEMs now just support their own products and community on their forums. 

For the most part, it's completely appropriate to ask Grbl-related questions here. Especially if it's a short succinct question that has shown some willingness to do and have done work on their own. 

But please don't expect to have a broad array of questions to be answered completely or at all. It's way too difficult to get any useful information from threads like these because the conversation tends to jump around everywhere. 

That said, if you have multiple questions, don't open a dozen issues either. It's best to prioritize the most important one or two and then pose them. You'll usually have the best luck getting the rest of your questions answered once a thread gets going. 

And also don't expect an issues thread to remain open if it is not part of a Grbl related issue. They will be closed but are allowed to continued.

closed 2017-08-06 by chamnit

gerritv, 2017-08-06:
@jahnj0584 I am not being rude, I am being direct. There are hundreds of people who see these emails as a result of a general 'can you help me?' issues that are not related to Grbl itself. They take time to read and even discard. I am at the point of not following anymore as the signal to noise ratio relating to Grbl itself has dropped to minimal. Worse it consumes Sonny's time, which we  wold rather see spent on the ARM version so that we can benefit from his skills and move to the next level of Grbl goodness.

As @KoljaWindeler said, try Google or Bing. I typed in the HuanYang VFD and on the first response screen there are 5 references to the manual. Most of the rest of your questions are answered on the wiki here. And even more answers in the Closed Issues. There have been multiple warnings about what not to do with EStop, and your suggestion is one of them. So while I just did your homework for you, you still have to do the same because I won't be providing the links.

KoljaWindeler, 2017-08-06:
You're right, let's help him:
http://bfy.tw/DEYt

Or let's make it even better
https://www.shapeoko.com/forum/viewtopic.php?f=21&t=5276

Harvie, 2020-11-23:
Now there seems to be quite elegant solution integrating Huanyang RS485 communication directly to GRBL, i would be happy if you can help with testing:
https://github.com/terjeio/grblHAL/issues/68