gnea/grbl #247: Axis co-ordinates not changing.

StuartB4, 2017-08-12:


chamnit, 2017-08-12:
You'll have to be more specific than a simple title to get any help.

closed 2017-08-12 by chamnit

gerritv, 2017-08-12:
Your UGS is way out of date, you need a newer version to work with Grbl 1.1.
Grbl Controller I think is no longer maintained, it will not work with Grbl 1.1

StuartB4, 2017-08-12:
All 3 programs work fine with GRBL 9, whats changed to make them stop working properly?

chamnit, 2017-08-12:
The comm protocol to include more data. As stated in the wiki and the front page. You need a v1.1 compatible sender.

StuartB4, 2017-08-12:
Ok, thanks chamnit. Just trying Candle 1.1.7 and it seems to work ok.

@gerritv:
"Your UGS is way out of date, you need a newer version to work with Grbl 1.1."

I have 1.0.9 as you can see from the top of the image I posted, thats the latest one I can find here:
https://winder.github.io/ugs_website/download

gerritv, 2017-08-12:
To be more explicit:
Your UGS version is way out of date. in 2015 Gnea/Grbl 1.1 did not exist so it won't support the new reporting format. You will need to ask on the UGS Github because I a) don't use UGS, b) can't make easy sense of what to download that would support 1.1 but I know they have a version that does.