gnea/grbl #248: Axis co-ordinates not changing.

StuartB4, 2017-08-12:
Sorry about that, pressed enter by mistake and it posted it.

I've just updated my UNO to grbl 1.1 and have a problem with the gcode sending programs I'm using.
I have Universal GCode Sender, Candle, and GRBLController. When moving the X, Y and Z axis nothing changes in the co-ordinate boxes. All 3 programs do the same. Also the virtual indicator in the programs does'nt move.
[image]
[image]
[image]

gerritv, 2017-08-12:
See Issue #247

StuartB4, 2017-08-12:
@gerritv: nothing there. That's the one i posted by mistake.

closed 2017-08-12 by chamnit

gerritv, 2017-08-12:
What I wrote there is gone:
Your UGS version is way out of date. in 2015 Gnea/Grbl 1.1 did not exist so it won't support the new reporting format. You will need to ask on that Github because I a) don't use UGS, b) can't make easy sense of what to download that would support 1.1

Candle has a branch for Grbl 1.1

Grbl Controller is abandoned.