gnea/grbl #249: Controller reboots on some G-Code files

makkarpov, 2017-08-12:
When streaming [this](https://mega.nz/#!D0RVEbrQ!KfJkKiZZG1QBvcI9aliIkFvqUpPiXdandxui2qAOH38) file to MCU running GRBL 1.1f it suddenly halts after reception of command on line 1023.

In console (UGS 2.0 nightly) it looks like this:
```
>>> X-22.919Y-65.224
ok
>>> X-22.969Y-65.125
ok
>>> X-23.035Y-65.052
ok
>>> X41.358Y-65.052

Grbl 1.1f ['$' for help]
```

In Candle v1.17 it looks like this (after that nothing happens, no error messages or something like that):
<details>
<summary>Screenshot</summary>

[image]

</details>
<hr />
Output of <code>$$</code> and <code>$#</code> commands is attached below:
<details>
  <summary>Some settings from the MCU</summary>

```
$0=10
$1=25
$2=0
$3=5
$4=0
$5=0
$6=0
$10=3
$11=0.010
$12=0.002
$13=0
$20=0
$21=0
$22=0
$23=0
$24=25.000
$25=500.000
$26=250
$27=1.000
$30=1000
$31=0
$32=0
$100=800.000
$101=800.000
$102=800.000
$110=5000.000
$111=5000.000
$112=500.000
$120=30.000
$121=30.000
$122=20.000
$130=200.000
$131=200.000
$132=200.000
ok
[G54:143.959,-38.721,-5.000]
[G55:0.000,0.000,0.000]
[G56:0.000,0.000,0.000]
[G57:0.000,0.000,0.000]
[G58:0.000,0.000,0.000]
[G59:0.000,0.000,0.000]
[G28:0.000,0.000,0.000]
[G30:0.000,0.000,0.000]
[G92:0.000,0.000,0.000]
[TLO:0.000]
[PRB:0.000,0.000,0.000:0]
ok
```
</details>

chamnit, 2017-08-12:
You are running at 66kHz step rates (step/mm * mm/min / 60 sec/min). Grbl supports up to 30kHz. Reduce your micro stepping so that your step/mm are around 200.

closed 2017-08-12 by chamnit

makkarpov, 2017-08-12:
But to reduce microstepping I must adjust hardware, right? (It seems that my hardware has no such adjustmens). If so, I better reduce maximum movement speed, anyway that fact that it completes almost everything and freezes almost on the end means that during normal operation these limits are not reached.